j7126/OctoPrint-Dashboard #170: Webcam Stream breaks when using Dashboard Plugin

anupchavda1, 2020-06-02:
Have a RPi 4 (16gb) and have been using it for some time with latest updated Octoprint. 
Have had the Webcam working consistently from day 1 (on the Control Tab as well as Octolapse.

Configuration on OctoPrint for the Webcam and Timelapse page: Enable Webcam is checked and Stream URL is `/webcam/?action=stream`

Installed Dashboard on 2 different Octoprint instances and it immediately breaks ALL webcam streams (Control Page, Dashboard Page, Octolapse etc). Uninstalling Dashboard brings the feed back.

Any ideas?

StefanCohen, 2020-06-02:
Thanks for the feedback. 

That seems strange. I have the same setup here and it works fine. Do you see any errors in the Octoprint log or in the javascript console of your browser?

StefanCohen, 2020-06-03:
@Willmac16 That seems unrelated as OPs issue was reported prior to the release of 1.13.0 that included multicam support. @CynanX Do you have any idea what could cause this issue?

anupchavda1, 2020-06-03:
It’s weird. It’s now working, BUT only on the dashboard. My “control” tab stays on the getting stream for a few mins and then shaves to unable to get stream

CynanX, 2020-06-03:
Afraid not.

The second camera is selected by default on the dashboard due to a quirk. There are icons on the top right of the stream where you can switch the streams... they are black and perhaps to faint to see on your system?

Willmac16, 2020-06-03:
Yup, I can faintly make them out now. Would it be possible to make a dark theme friendly version of them since the rest of the plugin works so well with the dark theme?

StefanCohen, 2020-06-03:
I’ll look at it.

Willmac16, 2020-06-03:
I think adding `filter: contrast(0);` to .cameraStreamIcon and slightly increasing the opacity does the trick.

StefanCohen, 2020-06-03:
Thanks. I'll look at the PR once I'm done with a small experiment.

StefanCohen, 2020-06-04:
That PR looks good. Merged and ready for v.1.14.0

j7126, 2020-09-29:
can this be closed @StefanCohen

closed 2020-09-29 by StefanCohen