j7126/OctoPrint-Dashboard #171: Dashboard plugin reporting "Can't get stats from DisplayLayerProgress."

msteele999, 2020-06-03:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
I have a fresh installation of OctoPrint 1.4.0 on Windows 10 Home using Python 2.7 32 bit. I have a stock install of 1.4.0 with only Dashboard 1.13.0 and DisplayLayerProgress 1.21.0 plugins installed. This issue started a couple of days ago where the dashboard would show all elements but have the 
"Can't get stats.

I have removed the Octoprint directory, uninstall Python, cleared out the octoprint AppData directories and cleared all temp files. I have cleared my browser cache and used four different web browsers, all with the same results.

Is it possible there is a directory I've missed or a registry entry I need to kill in order to clear out any old installation?

**To Reproduce**
1. Install Python
2. Install virtualenv
3. Install / Configure octoprint
4. Install Dashboard and displaylayerprogess plugins
5. Refresh the page.

**Expected behavior**
Dashboard elements reporting temps from printer

**Screenshots**
Attached
[image]
[image]

**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log
[octoprint.log](https://github.com/StefanCohen/OctoPrint-Dashboard/files/4722852/octoprint.log)

* Browser console log
[192.168.1.159.zip](https://github.com/StefanCohen/OctoPrint-Dashboard/files/4722856/192.168.1.159.zip)

* gcode file that can be used to reproduce the problem
N/A

**OctoPrint server and Plugins:**
 - Platform: [e.g. RPi 3b+, Intel x86] Intel x86
 - OS and version: [e.g. Raspbian Stretch, Ubuntu 18.10, Windows 10] Windows 10 Home
 - OctoPrint version: [e.g. OctoPrint 1.3.11] 1.4.0
 - OctoPi version: [e.g. OctoPi 0.16.0] N/A
 - Dashboard version [e.g. Dashboard 1.10.1] 1.13.0
 - DisplayLayerProgress version [e.g. 1.14.0] 1.21.0

**Desktop (please complete the following information):**
 - OS and version: [e.g. Windows 10]
 - Browser and version [e.g. chrome 77.0.3865, safari 12.1.2] Version 83.0.4103.61 (Official Build) (64-bit)

**Additional context**
Add any other context about the problem here.

msteele999, 2020-06-03:
I have confirmed that DisplayLayerProgress is working - see screen shots:

[image]

StefanCohen, 2020-06-03:
Thanks for the feedback. This has most likely nothing to do with DLP. Can you please check your javascript error console? This looks like a viewmodel binding problem.

msteele999, 2020-06-03:
I'm having trouble accessing my java console (I have it enabled - just not
seeing it yet).

Is there a minimum Java version you require?

msteele999, 2020-06-03:
here we go:

Less has finished and no sheets were loaded.
packed_core.js?7e471afe:14648 Starting dependency resolution...
packed_core.js?7e471afe:14747 Could not instantiate the following view
models due to unresolvable dependencies:
(anonymous) @ packed_core.js?7e471afe:14747
mightThrow @ packed_libs.js?3339d949:3535
process @ packed_libs.js?3339d949:3603
setTimeout (async)
(anonymous) @ packed_libs.js?3339d949:3641
fire @ packed_libs.js?3339d949:3269
fireWith @ packed_libs.js?3339d949:3399
fire @ packed_libs.js?3339d949:3407
fire @ packed_libs.js?3339d949:3269
fireWith @ packed_libs.js?3339d949:3399
ready @ packed_libs.js?3339d949:3879
completed @ packed_libs.js?3339d949:3889
packed_core.js?7e471afe:14749 dashboardViewModel (missing: gcodeViewModel,
enclosureViewModel )
(anonymous) @ packed_core.js?7e471afe:14749
Pn @ packed_libs.js?3339d949:11208
(anonymous) @ packed_libs.js?3339d949:11229
(anonymous) @ packed_core.js?7e471afe:14748
mightThrow @ packed_libs.js?3339d949:3535
process @ packed_libs.js?3339d949:3603
setTimeout (async)
(anonymous) @ packed_libs.js?3339d949:3641
fire @ packed_libs.js?3339d949:3269
fireWith @ packed_libs.js?3339d949:3399
fire @ packed_libs.js?3339d949:3407
fire @ packed_libs.js?3339d949:3269
fireWith @ packed_libs.js?3339d949:3399
ready @ packed_libs.js?3339d949:3879
completed @ packed_libs.js?3339d949:3889
packed_core.js?7e471afe:14758 ... dependency resolution done
packed_core.js?7e471afe:15068 Initial application setup done, connecting to
server...
packed_core.js?7e471afe:12694 Connected to the server
packed_core.js?7e471afe:15033 Finalizing application startup
packed_core.js?7e471afe:14908 Going to bind 34 view models...
packed_core.js?7e471afe:14961 Did not bind view model UsageViewModel to
target #wizard_plugin_tracking since it does not exist
packed_core.js?7e471afe:14961 Did not bind view model
SoftwareUpdateViewModel to target #softwareupdate_confirmation_dialog since
it does not exist
packed_core.js?7e471afe:14961 Did not bind view model
SoftwareUpdateViewModel to target #wizard_plugin_softwareupdate since it
does not exist
packed_core.js?7e471afe:3334 User msteele logged in
packed_core.js?7e471afe:15000 ... binding done
packed_core.js?7e471afe:15012

Looks like a dependency issue?

StefanCohen, 2020-06-03:
Thanks:

packed_core.js?7e471afe:14749 dashboardViewModel (missing: gcodeViewModel,
enclosureViewModel )

This indicates that you have disabled the gCode Visualizer in the octoprint settings. Please try enabling it.

msteele999, 2020-06-03:
Well there you go - who knew?

I disabled it so it would not show in the tabs as I use PrettyGcode viewer.
I will have to find another way of customizing what shows in the tab.

Working now!

StefanCohen, 2020-06-03:
Perfect. Closing

closed 2020-06-03 by StefanCohen