j7126/OctoPrint-Dashboard #172: element IDs for customization

Klammerson, 2020-06-03:
I would like to customize my dashboard but css is hard to apply since some elements have IDs others just have classes.

Some elements even have styles in the code itself.

StefanCohen, 2020-06-04:
Thanks for the feedback. I agree. It is time for a css cleanup. It's a bit messy due to all the different view modes and browser quirks so this will take some time.

Klammerson, 2020-06-04:
If you want, I'd be happy to assist!?

StefanCohen, 2020-06-04:
Sure. CSS is not my strong point. I'm currently working on a new feature that I want to complete but feel free to fork the project and make yourself familiar with the code while I finish the current feature. I'm all over the code at the moment so merging a PR would be quite messy until I'm done with this.

Klammerson, 2020-06-04:
thx. I will make some changes and post them here so you can review them.

I also found a typo in the dashboard_tab.jinja2 which might or might not affect to overall design.

StefanCohen, 2020-06-04:
Thanks. I'm done with the new feature so you may fork and pull the main branch as it is up to date and prepped for the next release. You can then create a pull request when you are ready.

Klammerson, 2020-06-05:
I'm sorry but I'm really new to plugin development. If I download your plugin and unzip it locally, make some changes, how can I test my changes locally?
I have setup a VM with OctoPrint running where I want to test it.

StefanCohen, 2020-06-05:
No problem. The best way to do it is to fork the project to your own github account. You do that using the button in the top right corner of this page. You then clone your own fork from github to your development environment using the git clone command. I usually clone it to `/home/pi/.octoprint/plugins` 
Then activate the virtual env using `source /home/pi/venv/bin/activate`

The plugin is now ready to be installed via python setup.py install. However, since we are still working on our plugin, it makes more sense to use python setup.py develop for now – this way the plugin becomes discoverable by OctoPrint, however we don’t have to reinstall it after any changes we will still do. We can have the octoprint dev plugin:install command do everything for us here, it will ensure to use the python binary belonging to your OctoPrint installation:

```
(venv) $ octoprint dev plugin:install
running develop
running egg_info
creating OctoPrint_HelloWorld.egg-info
[...]
Finished processing dependencies for OctoPrint-Dashboard==1.14.0
```
Restart OctoPrint. Your plugin should still be properly discovered. Make your changes, test them locally, push your changes to your github fork and finally create a pull request against my repository on github. I can then review the changes and merge them with the master branch.

It may sound complicated but it is really straight-forward once you get a hang of it. The Octoprint plugin tutorial is really helpful: https://docs.octoprint.org/en/master/plugins/gettingstarted.html

Klammerson, 2020-06-05:
> No problem. The best way to do it is to fork the project to your own github account. You do that using the button in the top right corner of this page. You then clone your own fork from github to [...]

successfully did that

> Then activate the virtual env using `source /home/pi/venv/bin/activate`

successfully did that

> The plugin is now ready to be installed via python setup.py install. However, since we are still working on our plugin, it makes more sense to use python setup.py develop for now – this way the [...]

runs but doesn't install the plugin

> Restart OctoPrint. Your plugin should still be properly discovered. Make your changes, test them locally, push your changes to your github fork and finally create a pull request against my [...]

StefanCohen, 2020-06-05:
Hmm. I think you need to run `octoprint dev plugin:install` from within the root folder of the cloned plugin. That should be the OctoPrint-Dashboard folder that contains setup.py

Klammerson, 2020-06-05:
that worked like a charm. I will now do my best to apply my ideas

StefanCohen, 2020-06-05:
Cool. I'll be travelling during the weekend but will try to support you when I can find the time so just ask away here.

StefanCohen, 2020-06-09:
I have manually gone over the PR to see what went wrong but couldn't find the reason so I manually pulled the typo fix and your class name updates. That seems to work fine so I will include at least that part in the upcoming v1.14 release.

j7126, 2020-09-30:
@StefanCohen what is the status on this, can it be closed?

StefanCohen, 2020-09-30:
Fixed by 1.15.0. Closing

closed 2020-09-30 by StefanCohen