j7126/OctoPrint-Dashboard #174: [Bug] Loose camera configuration

courtpl, 2020-06-04:
IIt seems a problem between the Dashboard and Multicam pluggin

In Multicam the button name are inverted and the camera configuration reset as defaut camera path for all camera.
In dashboard, The second camera is displayed instead of the default camera.

The problem appear after each reboot

StefanCohen, 2020-06-04:
Thanks for the feedback. I'll take a look.

StefanCohen, 2020-06-05:
@CynanX I found the problem. It was in the onBeforeBinding method.

It looked like this: 

`self.multicam_profiles(self.settingsViewModel.settings.plugins.multicam.multicam_profiles().reverse());`

That reversed the actual multicam plugin settings to be reversed while I assume that you intended the Dashboards internal self.multicam_profiles to be reversed. So I changed it to:

```
self.multicam_profiles(self.settingsViewModel.settings.plugins.multicam.multicam_profiles());
self.multicam_profiles.reverse();
```

Does that look right to you? It works fine in my tests and stops Dashboard from interfering with the multicam plugin settings. It still doesn't prevent the last cam stream from loading first in the Dashboard but I'll take a look at that next.

CynanX, 2020-06-05:
Sweet!

Coming from Java I thought it was immutable. Hmm.. need to look into Python unit tests 😀

CynanX, 2020-06-05:
PS. The last stream becoming active seems to be because the onClick method in the HTML seems to be processed upon load.

StefanCohen, 2020-06-05:
Ah. I can take a look at that. One thing that still bugs me is that the wrong camera button in the control tab is selected while the right (default) stream is loaded (after my fix). It could be related to the onClick event firing on the wrong viewModel of course.... Hmm. I need to debug this.

StefanCohen, 2020-06-05:
I tried removing `, click: function(){$parent.webcamState( $index()+1 );}` from the icon data binding and it still loads the last stream by default so it seems unrelated to the onClick event firing on load. Very strange.

courtpl, 2020-09-30:
Allways the same problen, even after installation of the last version 1.15.0.
Also after some minutes of usage, the server become totally busy.

I have reinstall the version 1.12.2

courtpl, 2020-10-01:
I have installed the las 1.15 version, but the problem is always the same.
It appear, the server became instable after some minutes of usage.

I have perform a rollback to the version 1.12.2, back to normal.

gneu42, 2020-10-04:
Hi,
It seems that I have the same problem.
I upgrade multicam and dashboard plugin from 1.11.0 to 1.15.0, Everything looks ok. The next day I restart the printer, and no more second camera, but could access them on my browser. It and the camera seems to have been inverted in octoprint. My original default camera was a raspicam and my second was USBC920, now the default is USB920 and the second one is raspicam. In multicam, both show the same URL : http://192.168.1.183/webcam/?action=stream. No more webcamd2 and the name are inverted.
I can't change the URL of the first one wich is now USB920. I would like to rollback to 1.12.2, but don't have it since I went from 1.11 to 1.15.
Does anyone knows how I can get 1.12.2 on my PI ?

Regards Gérard

Willmac16, 2020-10-04:
Hi there. I am actively working on a fix/replacement of the multiple camera functionality, and I believe you can install an earlier version by using the install url with "master" replaced with the release number.

StefanCohen, 2020-10-04:
There is a wiki page describing how to install earlier versions here: https://github.com/j7126/OctoPrint-Dashboard/wiki/Install-an-older-version-of-the-plugin

gneu42, 2020-10-04:
Thank you for the info. I rollback to 1.13.0. And it seems better than 1.15.0.
- Both cameras are working ok.
- The order of the cameras on the multicam setup and control tab are back to normal.
- The button on my small touch screen using touchUI have an icon showing a camera, while with 1.15.0 they were just white squares.
I'll wait until tomorrow to see if it is stable.

Regards
Gérard

Willmac16, 2020-10-05:
I've pushed a commit to the development branch that adds multiple camera support to dashboard without using Multicam.

gneu42, 2020-10-05:
After the rollback of yesterday to 1.13.0, everythink looked ok, but this morning when I restart the printer, the camera were inverted in multicam, and both pointing to webcamd2 !!!

j7126, 2020-10-06:
this will be fixed in the next release

j7126, 2020-10-07:
should be fixed in 1.15.1

closed 2020-10-07 by j7126

gneu42, 2020-10-08:
I installed 1.15.1, and the camera selector buttons are gone. Only the first camera is displayed !!!

j7126, 2020-10-08:
you have to add the cameras in the dashboard settings, it no longer gets them from the multicam plugin @gneu42

gneu42, 2020-10-08:
Ok, Now it is working well.

Great job.

Thank you