j7126/OctoPrint-Dashboard #176: ETL and ETA way off

vannovv, 2020-06-05:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
Dashboard reports "bad accuracy" and ETL is way off.  Estimated total print time shows 9:52:19 and ETL shows 17:19:22 left which is causing the ETA time to be calculated incorrectly 14:17:19 the following day for example.  

**To Reproduce**
Steps to reproduce the behavior:
Kick off print job 

**Expected behavior**
Expect ETL and ETA to reflect closer to actual print times
**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log
[octoprint.log](https://github.com/StefanCohen/OctoPrint-Dashboard/files/4737970/octoprint.log)

* Browser console log
* gcode file that can be used to reproduce the problem

[CE3PRO_single_gcode.txt](https://github.com/StefanCohen/OctoPrint-Dashboard/files/4737979/CE3PRO_single_gcode.txt)

**OctoPrint server and Plugins:**
 - Platform: [e.g. RPi 3b+, Intel x86] 
RPI 3
 - OS and version: [e.g. Raspbian Stretch, Ubuntu 18.10, Windows 10]
 Raspbian/Octoprint image
 - OctoPrint version: [e.g. OctoPrint 1.3.11] 
Version 1.4.0

 - OctoPi version: [e.g. OctoPi 0.16.0]
 - Dashboard version [e.g. Dashboard 1.10.1] 
1.14.0
 - DisplayLayerProgress version [e.g. 1.14.0]
1.21.0
**Desktop (please complete the following information):**
 - OS and version: [e.g. Windows 10] Mac 10.15.4
 - Browser and version [e.g. chrome 77.0.3865, safari 12.1.2] 
Safari Version 13.1

**Additional context**
Add any other context about the problem here.

StefanCohen, 2020-06-05:
Thanks for the feedback. The estimated print time is provided by OctoPrint and simply displayed by the Dashboard. You will typically find it in the sidebar. It is notoriously inaccurate but can be improved by using the Print Time Genius plugin. Dashboard will in that case pick up the improved version. The ETA is calculated by the dashboard based on the estimated print time so the accuracy will depend on the source.

vannovv, 2020-06-06:
Thanks for the response.  I have PrintTimeGenius install and up to the point when I installed the dashboard (which I really like by way) the estimates were pretty close to actual time.  Now full disclosure I did install some other plugin as well when I installed the dashboard.  Maybe I should delete and reinstall PrintTimeGenius and see how that goes.  Printing right now it will be late tonight before I and try this.

StefanCohen, 2020-06-06:
Ok. Closing this issue as there is not mich I can do about it.

closed 2020-06-06 by StefanCohen