j7126/OctoPrint-Dashboard #177: Clear cached data on Jo

StefanCohen, 2020-06-06:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log
* Browser console log
* gcode file that can be used to reproduce the problem

**OctoPrint server and Plugins:**
 - Platform: [e.g. RPi 3b+, Intel x86]
 - OS and version: [e.g. Raspbian Stretch, Ubuntu 18.10, Windows 10]
 - OctoPrint version: [e.g. OctoPrint 1.3.11]
 - OctoPi version: [e.g. OctoPi 0.16.0]
 - Dashboard version [e.g. Dashboard 1.10.1]
 - DisplayLayerProgress version [e.g. 1.14.0]

**Desktop (please complete the following information):**
 - OS and version: [e.g. Windows 10]
 - Browser and version [e.g. chrome 77.0.3865, safari 12.1.2]

**Additional context**
Add any other context about the problem here.

fixyourlan, 2020-09-11:
This should be cleared when a new print is started, not at the end, so you can still see the stats of the last print, as it is now its cleared as soon as its finished and you are unable to see how long it took to print. Before it was just leaving the old info even after the new print was started, I think it was until the first layer was completed, which was confusing, now its cleared too soon for me!

StefanCohen, 2020-09-27:
@j7126 @Willmac16 The comment from @fixyourlan is interesting as it in conflict with my original design vision. See #145 for context. I created this ticket to go in the other direction and clear everything on print end/cancel. I mentioned this on our call earlier today but I think this puts it into context a bit better.

j7126, 2020-09-30:
I think maybe it could be a setting to choose if it gets cleared at the start or the end for each widget. Additionally that setting would control if that widget is still shown when no print is running, this way for example you could still see they layer times graph  form the previous print after it is done (which currently even though it is cleared at print start, it is not visible until print start anyway).

j7126, 2021-02-27:
As per my previous comment, I have implemented a setting for each widget which controls if a widget is hidden when not currently printing or not. 

I have implemented a setting that controls if the data is cleared on print start, print end or never for the printer message, layer graph, filament, and feed-rate widgets. 

These settings are located in the new visibility settings modal for each widget, this modal also contains the 'show in full-screen' checkbox, and the button for this modal is located where that checkbox was.

closed 2021-03-15 by j7126