j7126/OctoPrint-Dashboard #182: Fullscreen Views Not Correct Z-Index

jneilliii, 2020-06-11:
I received a report of an issue on my OctoPrint-Widescreen plugin that adds a right sidebar that left sidebar panels can be moved into. When this happens it causes an overlap issue.

I could patch my plugin to override your styles, but I think it would be more important for you to patch this in a way that will not cause this with any plugin that also modifies the UI. 

In order to do that you need to add a `z-index: 1000` or some other high number to your css definitions for the dashboard-full class [here](https://github.com/StefanCohen/OctoPrint-Dashboard/blob/277da3b4b19a9b5aabfb0e016fea77d69e1a328b/octoprint_dashboard/static/css/dashboard.css#L505).

StefanCohen, 2020-06-11:
Thanks for the heads-up. Will be fixed in v1.14. Did you by the way see my comment regarding the correction of the css typo? It will also be included in v1.14.

jneilliii, 2020-06-11:
I did. I just pushed an update to my plugin that should handle both cases.

gdachs, 2020-08-04:
I have a related problem
[image]
I use version 1.14 of your plugin

jneilliii, 2020-08-04:
@StefanCohen I just checked your css with current master branch and there is no additional z-index properties assigned as suggested. I assume that's why this issue is still open and didn't make it into version 1.14? BTW, you forgot to add the release here in github for that version 1.14, so there's a chance if you push any additional updates to master prior to adding the release some users that installed anew won't get their install updated.

Willmac16, 2020-09-30:
The fix should now be in the code.

closed 2020-09-30 by Willmac16