j7126/OctoPrint-Dashboard #185: Command page has default command that clutters Octoprint logs

bjhiltbrand, 2020-06-29:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
When the dashboard plugin is installed, a default command is setup in the new command widgets section. This default command simply echoes "9V" to the logs. Please see screenshots.

**Expected behavior**
I don't know if this default command is required, but I did delete it and it doesn't seem to have any impact on the rest of Octoprint, and has stopped writing out to the logs. If it's not required, should it be there in the first place?

**Screenshots**
[image]
[image]

**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log
* Browser console log
* gcode file that can be used to reproduce the problem

**OctoPrint server and Plugins:**
 - Platform: RPi 3b+
 - OS and version: Octopi
 - OctoPrint version: OctoPrint 14
 - OctoPi version: OctoPi 0.17.0
 - Dashboard version Dashboard 1.14

**Desktop (please complete the following information):**
 - OS and version: Windows 10
 - Browser and version: chrome 83.0.4103.116

bjhiltbrand, 2020-06-29:
Looks like I spoke too soon. I deleted that command widget, but now I just have a number of lines in my log that look like this:

2020-06-29 11:50:23,702 - octoprint.plugins.dashboard - INFO - cmd_results: 

So, same as before, just without the "V9" appended to the end.
Any help on this would be appreciated. Thanks.

StefanCohen, 2020-07-08:
The latest official release of Dashboard is 1.13.0. It looks like you have manually installed v1.14. That version has not yet been released to the OP plugin repo and still has some open issues. This is one of them so I will close this ticket as it has been fixed in the not yet released 1.14.0.

bjhiltbrand, 2020-07-08:
I simply installed the plugin from the OP plugin manager. I just uninstalled it and re-installed it, and it installed v1.14 again. How do I get back down to 1.13?

StefanCohen, 2020-07-08:
Hmmm. Something is strange. The latest release I made on github was 1.13.0. I will have to investigate how 1.14.0 has made its way to the plugin repo.

[image]

bjhiltbrand, 2020-07-08:
The only reference to 1.14.0 that I can find is in your wiki, but every other reference, including on the OP repository webpage, says the latest version is 1.13.0. I have no clue how it got a hold of the newer version.

bjhiltbrand, 2020-07-08:
setup.py from the downloaded zip from the master branch has 1.14.0 listed on line 19 as the plugin version.

StefanCohen, 2020-07-08:
Ok. That means that I messed something up during a merge or rollback. I will try to get 1.14.0 out of the door to fix this mess.

bjhiltbrand, 2020-07-08:
Ok. Thank you. :)

Taomyn, 2020-08-12:
Hi, I have 1.14.0 installed for some time and in diagnosing another Pi system problem today noticed this clogging up my syslog:

```
Aug 11 10:16:27 otto octoprint[13487]: 2020-08-11 10:16:27,237 - octoprint.plugins.dashboard - INFO - cmd_results:
Aug 11 10:16:30 otto octoprint[13487]: 2020-08-11 10:16:30,245 - octoprint.plugins.dashboard - INFO - cmd_results:
Aug 11 10:16:33 otto octoprint[13487]: 2020-08-11 10:16:33,250 - octoprint.plugins.dashboard - INFO - cmd_results:
Aug 11 10:16:36 otto octoprint[13487]: 2020-08-11 10:16:36,260 - octoprint.plugins.dashboard - INFO - cmd_results:
Aug 11 10:16:39 otto octoprint[13487]: 2020-08-11 10:16:39,269 - octoprint.plugins.dashboard - INFO - cmd_results:
Aug 11 10:16:42 otto octoprint[13487]: 2020-08-11 10:16:42,278 - octoprint.plugins.dashboard - INFO - cmd_results:
Aug 11 10:16:45 otto octoprint[13487]: 2020-08-11 10:16:45,284 - octoprint.plugins.dashboard - INFO - cmd_results:
Aug 11 10:16:48 otto octoprint[13487]: 2020-08-11 10:16:48,289 - octoprint.plugins.dashboard - INFO - cmd_results:
Aug 11 10:16:51 otto octoprint[13487]: 2020-08-11 10:16:51,295 - octoprint.plugins.dashboard - INFO - cmd_results:
Aug 11 10:16:54 otto octoprint[13487]: 2020-08-11 10:16:54,301 - octoprint.plugins.dashboard - INFO - cmd_results:
Aug 11 10:16:57 otto octoprint[13487]: 2020-08-11 10:16:57,307 - octoprint.plugins.dashboard - INFO - cmd_results:

```

How do I stop it or revert back to 1.13 before it wears another microSD card as this is the second one in the past 4 months? This is what happens when I re-install it:

```
Installing plugin...
Installing plugin "OctoPrint-Dashboard" from https://github.com/StefanCohen/OctoPrint-Dashboard/archive/master.zip...
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmpn4yzp6t2/OctoPrint-Dashboard-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpn4yzp6t2/OctoPrint-Dashboard-master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Dashboard==1.14.0) (1.4.2)
Requirement already satisfied: psutil in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint-Dashboard==1.14.0) (5.7.2)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.06.22 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (2020.6.22)
Requirement already satisfied: markdown<3.2,>=3.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (3.1.1)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.16.3)
Requirement already satisfied: Click<8,>=7.1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (7.1.2)
Requirement already satisfied: requests<3,>=2.23.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (2.24.0)
Requirement already satisfied: tornado==5.1.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (5.1.1)
Requirement already satisfied: sarge==0.1.5post0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.1.5.post0)
Requirement already satisfied: cachelib<1,>=0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.1.1)
Requirement already satisfied: wrapt<2,>=1.12.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.12.1)
Requirement already satisfied: OctoPrint-FileCheck>=2020.07.06 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (2020.8.7)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.5.0)
Requirement already satisfied: frozendict<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.2)
Requirement already satisfied: websocket-client<1,>=0.57 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.57.0)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.5.0.1)
Requirement already satisfied: rsa==4.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (4.0)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (3.4)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.4)
Requirement already satisfied: emoji<1,>=0.5.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (0.5.4)
Requirement already satisfied: werkzeug<2,>=1.0.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.0.1)
Requirement already satisfied: Flask-Babel<2,>=1.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (1.0.0)
Requirement already satisfied: regex!=2018.11.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoPrint-Dashboard==1.14.0) (2020.6.8)
[... cut ...]
Building wheel for OctoPrint-Dashboard (setup.py): finished with status 'done'
Created wheel for OctoPrint-Dashboard: filename=OctoPrint_Dashboard-1.14.0-py3-none-any.whl size=178840 sha256=f4fcf281e9a7d181ac656d2d48d301953f95b7bfbc00859820c07e0fc8b05315
Stored in directory: /tmp/pip-ephem-wheel-cache-h2bhvckm/wheels/04/30/14/9263785282cc3f492a27f06ab29f9638883535821163b61c9d
Successfully built OctoPrint-Dashboard
Installing collected packages: OctoPrint-Dashboard
Successfully installed OctoPrint-Dashboard-1.14.0
Done!
```

j7126, 2020-09-28:
this will be fixed in 1.15

Willmac16, 2020-09-30:
The plugin now shouldn't log the results of command widgets.

closed 2020-09-30 by Willmac16