j7126/OctoPrint-Dashboard #186: Layer progress indicator does not update on large files

KenLucke, 2020-07-07:
**Describe the bug**
Layer progress thermometer does not update on large files.

**To Reproduce**
Steps to reproduce the behavior:
1. Load a large file (I have not yet pinned down the size exact breaking point between working and non-working.  Example file below is 375 MB)
2. Start print.

**Expected behavior**
Progress of layer indicator should represent the rough % of the layer done, as it does correctly with smaller and medium sized files.  It remains "0%" (although showing 100%, full progress indicator) throughout the whole layer, all layers

**Screenshots**
video link (github does not allow videos): https://www.dropbox.com/s/nsyuhz5tf5q7oie/screencast%202020-07-07%2014-44-28.mp4

**Files (please attach the following Files or screenshots when applicable):**

**OctoPrint server and Plugins:**
 - Platform: RPi 4b 4GB
 - OS and version: Linux  4.19.118-v7
 - OctoPrint version: Version 1.4.0
 - OctoPi version: 0.10.0
 - Dashboard version 1.13.0
 - DisplayLayerProgress version 1.22.1

**Desktop (please complete the following information):**
 - OS and version: Mac OS X 10.15.5
 - Browser and version both Firefox & Safari

StefanCohen, 2020-07-08:
Hi and thanks for the feedback. The dashboard depends on the GCode Viewer tab for the stats. What do you see on that tab when this happens? Do you see a warning like: "You've selected for printing which has a size of XMB. Depending on your machine this might be too large for rendering and cause your browser to become unresponsive or crash."? If that's the case, you should try to increase the size threshold in the OctoPrint settings (asuming you have enough available memory).

KenLucke, 2020-07-08:
I have had the Dashboard do its thing properly, even when the GCODE viewer tab won't work because of the filesize (which itself is a PITA, but I've learned that there's not much I can do about it, a it's apparently a limitation with the  browser ability to handle that large a script chunk).  

However, that being said, I do have my gcode handleable size set WAY up past recommended (600mb) from when I attempted to overcome that issue (not that it helps, it still gags the browser).  So, if that's the case,  looks as if there's not going to be a solution.

StefanCohen, 2020-07-10:
Ok. That is the problem then. I may end up trying to re-implement the gcode parsing from the gcode viewer tab in the backend but I fear that it will be problematic.

closed 2020-07-10 by StefanCohen

KenLucke, 2020-07-10:
> Ok. That is the problem then. I may end up trying to re-implement the gcode parsing from the gcode viewer tab in the backend but I fear that it will be problematic.

OK, thanks anyway.  I'll just live with it :)

KenLucke, 2020-07-10:
Just had a thought... (not perfect, but my thoughts rarely are :) )...

The layer time indicator still seems moderately accurate, even on large files, as it is apparently generated by an average of the last few layers, rather than relying on parsing the gcode.

  Could you not take that estimated time (even though it may be less accurate) and use the elapsed time on a layer vs that estimate to generate an [Approximate] % done figure?  

[image]