j7126/OctoPrint-Dashboard #190: Printer message display truncated (beginning and end)

neowolfwitch, 2020-07-27:
The printer status message (Msg:....) is truncated. While the info is available elsewhere in Dashboard, it's still an annoyance.

For example- my printer displays layer progress. The printer message should display as:
Msg: 10% L=33/327

Instead it displays as:
Msg: 0% L=33/32

It seems like the first and last character are being truncated, but sometimes two digits are truncated from the %, so it will look like:
Msg: % L=29/32

I'm attaching a screen shot...
[image]

OctoPrint 1.4.0 running on OctoPi 0.17.0

Here are all the plugins/versions I am running:
Access Anywhere - The Spaghetti Detective (1.3.2)
Bed Visualizer (0.1.13)
Dashboard (1.13.0)
DisplayLayerProgress Plugin (1.22.1)
Enclosure Plugin (4.13.1

Computer is MacOS Catalina running Chrome browser.

StefanCohen, 2020-09-07:
Thanks for the feedback. I will have to investigate this.

StefanCohen, 2020-09-30:
It is reproducable by sending an M117 message that contains any space, "M", "1" or "7" character in either the beginning or end of the message.

The bug is in __init__.py in process_gcode here:

`self.printer_message = cmd.strip("M117 ")`

fix with: 

`self.printer_message = cmd.replace("M117 ", "")`

or something similar. strip() truncates each listed character from the head and tail of the source string while replace() replaces each substring occurrence.

@j7126 @Willmac16 Will you take a look at this?

j7126, 2020-09-30:
will do

j7126, 2020-09-30:
ready for next release

closed 2020-09-30 by j7126