j7126/OctoPrint-Dashboard #191: CPU temp issue

clintkoglin, 2020-07-30:
It seems like something may have changed in the new version of octoprint. I noticed today that the CPU temp reads 0c constantly.

akoylu, 2020-08-04:
It happens on my setup as well.

clintkoglin, 2020-08-04:
I did a full reinstall of octoprint and now it seems to be working again. No clue what happened or changed

akoylu, 2020-08-04:
My actual problem was; somehow I received 1.14 package and could not get rid of it. I renamed `octoprint` folder to force reinstall from scratch which didn't help either. Then I saw URL section in Plugin Manager and forced to install version 1.13. That time I was successful.

Do you know how to remove any plugin? I suppose octoprint is caching somehow.

englolo, 2020-08-07:
I had same problem on my Rpi3b+
solved with `changing code
```
        if "cpu-thermal" in thermal: #RPi
            self.cpu_temp = int(round((thermal["cpu-thermal"][0][1]))) 
```
to
```
        if "cpu_thermal" in thermal: #RPi
            self.cpu_temp = int(round((thermal["cpu_thermal"][0][1])))
```

mikulik86, 2020-08-07:
On Rpi4B showing CPU Temp 0.

DorjeDorf, 2020-08-09:
Same here on Pi4b. 
psutil shows temp if ran from a manual command.

turbatus, 2020-08-10:
Same here! Navbar temp shows it correctly, the dashboard show always 0 degrees. RPI3B+

jacksterz, 2020-08-14:
I'm also having the same issue after the recent Octoprint update (v 1.4.2) with my Pi 4. My Dashboard version is 1.14.0. 

I wouldn't wat to reinstall Octoprint just to get this working, is there a fix for the Pi 4 I could try?

KRUP85, 2020-08-17:
Same here Pi4b. Temp 0. I really don't want to reinstall the whole Octo ...

jdpg2, 2020-08-20:
Same issue here Pi4B fresh Octoprint install. Using Resource Monitor plugin until this is fixed to view temps

girrrrrrr2, 2020-08-31:
Can this fix be pushed to fix the whole plugin?
It would help a lot of people who might not be comfortable editing their files.

Also fix confirmed working on my side .

mikulik86, 2020-09-01:
> I have install Dashboard 1.13.0
> Raspberry Pi3b+ ,OctoPrint; 1.4.2, OctoPi 0.17 , with latest Raspbian updates.
> edited [ **init**.py ](https://github.com/StefanCohen/OctoPrint-Dashboard/blob/1.13.0/octoprint_dashboard/__init__.py#L32-L33)
> line:32,33
> ` if "cpu-thermal" in thermal: #RPi self.cpu_temp = int(round((thermal["cpu-thermal"][0][1])))`
> to
> ` if "cpu_thermal" in thermal: #RPi self.cpu_temp = int(round((thermal["cpu_thermal"][0][1])))`
> 
> This working for me. For some reason v1.14.0 at all doesn't work for me. Not getting any data from pi.
> Im using python3 as default.

Worked for me too. (python2.7)

spanzetta, 2020-09-07:
Same problem here
PI3 A+ with OctoPrint 1.4.2 Python 2.7.16 OctoPi 0.17.0
CPU always at 0°

StefanCohen, 2020-09-07:
It looks like a psutil recently has started to report the temp using "cpu_thermal" instead of "cpu-thermal". That's the draw-back of having to rely on string parsing I guess. I will have to take a look at alternatives that still works across different kinds of hosts in some generic way.

spanzetta, 2020-09-07:
I did the manual fix and it works!
Thanks

KRUP85, 2020-09-07:
> > I have install Dashboard 1.13.0
> > Raspberry Pi3b+ ,OctoPrint; 1.4.2, OctoPi 0.17 , with latest Raspbian updates.
> > edited [ **init**.py ](https://github.com/StefanCohen/OctoPrint-Dashboard/blob/1.13.0/octoprint_dashboard/__init__.py#L32-L33)
> > line:32,33
> > ` if "cpu-thermal" in thermal: #RPi self.cpu_temp = int(round((thermal["cpu-thermal"][0][1])))`
> > to
> > ` if "cpu_thermal" in thermal: #RPi self.cpu_temp = int(round((thermal["cpu_thermal"][0][1])))`
> > This working for me. For some reason v1.14.0 at all doesn't work for me. Not getting any data from pi.
> > Im using python3 as default.
> 
> I'm using a raspberry pi 4 with octoprint 1.32, octopi 0.17 and also the latest updates. I am, however, using python 2.7 if that makes a difference.
> Your fix worked for me using dashboard 1.14.0.
> For those looking, the path in my case was `~/oprint/lib/python2.7/site-packages/octoprint_dashboard/__init__.py`

This fix works. Perfect. Many thanks.

girrrrrrr2, 2020-09-09:
> Hi all,
> 
> I have been searching for the plugin rep for hours and without success... Could someone give me the path to the init.py to modify on the RPI?
> Thanks for your help

Type this into your terminal
sudo nano ~/oprint/lib/python2.7/site-packages/octoprint_dashboard/__init__.py

Then change the lines mentioned above.

LordDarkneo, 2020-09-09:
> Type this into your terminal
> sudo nano ~/oprint/lib/python2.7/site-packages/octoprint_dashboard/**init**.py
>
> Then change the lines mentioned above.

That was my first try, but the file does not exist (so nano opens an empty file). when using the  ~/oprint/lib/python2.7/site-packages/octoprint_dashboard/__init__.py, I got no more information on the RPI proc data (neither temp, nor %usage, nor any info:
[image]

girrrrrrr2, 2020-09-09:
> That was my first try, but the file does not exist (so nano opens an empty file). when using the ~/oprint/lib/python2.7/site-packages/octoprint_dashboard/**init**.py, I got no more information on [...]

You have to restart octoprint for it to take effect.

Might need to reboot the whole pi i forget

LordDarkneo, 2020-09-09:
> You have to restart octoprint for it to take effect.
>
> Might need to reboot the whole pi i forget

Sudo reboot now saved my day :) Thanks for it ;)

morphias2004, 2020-09-13:
Same issue on a Pi4B 4G with 0.17 and 1.4.2.

Happened after I updated to Python 3.7 via the script posted on the blog the other day.

The above edit of the __init__.py did not fix mine. 😢

clintkoglin, 2020-09-13:
Also happened to me again when I did the upgrade to python 3.7 via octoprint. I'm guessing syntax changed in python 3?

Moskito99, 2020-09-14:
Same problem here.
Since Python3 all numbers are 0.

clintkoglin, 2020-09-14:
Does anyone know where the files for plug-ins are under the new python? They are not where they were worn python 2.7

haplm, 2020-09-21:
> I have install Dashboard 1.13.0
> Raspberry Pi3b+ ,OctoPrint; 1.4.2, OctoPi 0.17 , with latest Raspbian updates.
> edited [ **init**.py [...]

This has to be due to the updates to the latest Raspbian. I'm running Raspberry Pi 4B, OctoPrint 1.4.2, OctoPi 0.17, and my psutil still returns values with dash:

```>>> import psutil
>>> thermal = psutil.sensors_temperatures(fahrenheit=False)
>>> print(thermal)
{'cpu-thermal': [shwtemp(label='', current=53.556, high=None, critical=None)]}
```

But as I mention below, it doesn't mean that it works :-)

> This working for me. For some reason v1.14.0 at all doesn't work for me. Not getting any data from pi.
> Im using python3 as default.

I can confirm that 1.14.0 doesn't work, and the stats are not getting populated

fractalsauce, 2020-09-24:
> I'm using a raspberry pi 4 with octoprint 1.32, octopi 0.17 and also the latest updates. I am, however, using python 2.7 if that makes a difference.
> Your fix worked for me using dashboard 1.14.0. [...]

Can confirm that this fix works on  OctoPrint 1.4.2 | Python 3.7.3 | OctoPi 0.17.0 using version 1.13.0.

The path in my case was `~/oprint/lib/python3.7/site-packages/octoprint_dashboard/__init__.py`

I tried the fix with version 1.14.0 on my setup and none of the RPi info worked. CPU clock speed, CPU% usage, and CPU temp all showed 0.

Seeing as this plugin has been abandoned and OctoPrint is making the switch to Python 3, I'm really hoping someone will pick up this project and make the required fixes going forward. Thanks to those that found a temporary solution!

StefanCohen, 2020-09-24:
Hmm. Well. Not entirely abandoned. I'm working on moving ownership to new maintainers as I don't have the bandwidth to do it myself anymore. Sorry for the delay but I simply don't have the time to do this properly anymore.

fractalsauce, 2020-09-24:
> 
> 
> Hmm. Well. Not entirely abandoned. I'm working on moving ownership to new maintainers as I don't have the bandwidth to do it myself anymore. Sorry for the delay but I simply don't have the time to do this properly anymore.

Oh ok awesome! I was just referring to the announcement from OctoPrint, but glad to see you're working on finding a home for it. If I knew anything about coding Python I'd help if I could, but I know my way around CLI and can copy/past and follow instructions but that's about it :p

j7126, 2020-09-27:
Merged Pull request #196
Will be in 1.15

j7126, 2020-09-28:
It seems that in 1.14 there was an additional issue with the command widgets failing that caused send_notifications to break.

j7126, 2020-09-29:
Fixed for 1.15.0

closed 2020-09-29 by j7126

morphias2004, 2020-10-11:
@j7126 

Yay! You lads fixed it - temps are back. Thank-you!  👍