j7126/OctoPrint-Dashboard #192: Webcam windows doesn't resize in smaller screens when in full screen mode / full browser

mmoraes80, 2020-07-31:
**Describe the bug**
When accessing OctoPrint on a tablet the webcam window doesn't properly resize and covers part of the dashboard panel making it impossible to access the Camera, Full Screen,  Full Browser Window buttons, and some of the icons.

**To Reproduce**
Open Octoprint using a tablet and switch it fo full-screen mode.

**Expected behavior**
The camera panel should resize itself.

**Screenshots**
Not available at this moment.

**Desktop (please complete the following information):**
 - Fire HD7 tablet
 - Silk Browser (chrome based browser)

DylanCDuke, 2020-08-06:
I second this. 

Seen on desktop, and I've tried changing the camera resolution and aspect ratio on my /boot/octopi.txt. Mine isn't covering the dashboard information, but is instead running off the bottom of the screen.

[image]

j7126, 2020-10-08:
I am working on this. should be ready in a couple weeks

j7126, 2020-10-23:
Fixed, these screenshots were taken on my phone

[image]
[image]
Should be ready for next release

closed 2020-11-09 by j7126