j7126/OctoPrint-Dashboard #193: Highly precise print height i.e. too many decimals

Taomyn, 2020-08-18:
**Describe the bug**
Highly precise print height

**To Reproduce**
Steps to reproduce the behavior:
No idea really have attached the gcode I was printing

**Expected behavior**
A few less decimal points maybe?

**Screenshots**
[image]

**Files (please attach the following Files or screenshots when applicable):**
[CE3PRO_IRIS_BOX_RM.gcode.txt](https://github.com/StefanCohen/OctoPrint-Dashboard/files/5090498/CE3PRO_IRIS_BOX_RM.gcode.txt)

**OctoPrint server and Plugins:**
 - Platform: RPi-4
 - OctoPrint version: 1.4.2
 - Dashboard version: 1.14.0 (have tried getting back to 1.13.0 but the newer one keeps coming back, there's another issue logged about this)
 - DisplayLayerProgress version: 1.23.0

OllisGit, 2020-08-30:
fyi: I created a patch for this issue. 
If you can't wait for the next release of the Dashboard-Plugin, just modify the file by your own in your installation.

Taomyn, 2020-08-31:
I confirm fixed after manual modification of file, thanks.

closed 2020-08-31 by Taomyn