j7126/OctoPrint-Dashboard #197: Warning! Can't get stats from DisplayLayerProgress. Is it installed?

Makhaira, 2020-09-03:
Despite having the relevant plugins installed, Dashboard simply displays the message "Warning! Can't get stats from DisplayLayerProgress. Is it installed?" both during and outside of prints. I have it working successfully on another Octoprint instance

[image]
[octoprint.log](https://github.com/StefanCohen/OctoPrint-Dashboard/files/5168270/octoprint.log)

Octoprint Version 1.4.2
Dashboard (1.14.0)
DisplayLayerProgress Plugin (1.23.2)
Multi Line Terminal (0.1.3)
OctoKlipper (0.2.5)
Terminal Commands Extended (0.1.5)
Themeify (1.2.2)

Despite reinstalling both Dashboard and DisplayLayerProgress I cannot get Dashboard to stop displaying this error, which renders the rest of Dashboard largely useless. Any help would be gratefully received.

StefanCohen, 2020-09-07:
It looks like you may have something else going on that causes a Javascript error. That in turn causes the error message to be displayed even if it has nothing to do with your problem. Please take a look at your JS error console in the browser and report here if you find any errors.

Makhaira, 2020-09-07:
Thanks I'll check that out tonight.

Makhaira, 2020-09-07:
[image]

This is what I find @StefanCohen

chilluniverse, 2020-09-07:
@Makhaira Did you deactivate the "GCode Viewer" Plugin?

Had the same issue and as you and after I saw this screenshot I realised that I deactivated the GCode Viewer.
So I activated it again and this fixed the problem.

> [image]
> 
> This is what I find @StefanCohen

Makhaira, 2020-09-07:
100% correct! Many thanks. FAQ worthy?

StefanCohen, 2020-09-07:
Great. It has been in the Help section on the wiki for quite some time: https://github.com/StefanCohen/OctoPrint-Dashboard/wiki/Help#the-dashboard-fails-to-load-with-the-dashboardviewmodel-missing-gcodeviewmodel-error-message

Do you have a suggestion on how to improve it?

closed 2020-09-07 by StefanCohen

Makhaira, 2020-09-07:
I think the missing info was I never knew to look in the console for that
error message. There's a section somewhere that said you'd get this if
displaylayerprogress wasn't installed, it would maybe be worth linking the
two?

Thanks again for the help.

StefanCohen, 2020-09-07:
That's a good idea. I'll do it right away. Thanks!