j7126/OctoPrint-Dashboard #198: Wrong total height (eg 2.0000000447mm instead of 2.00mm)

spanzetta, 2020-09-03:
the total height shows a lot of decimals (wrong)

See screenshot below

It happen always

 - Platform: **RPi 3A+**
 - OctoPrint version: **1.4.2**
 - Dashboard version: **1.14**
 - DisplayLayerProgress version: **1.23.2**

 - OS and version: **Windows 10**
 - Browser and version: **Firefox 80**

[image]

spanzetta, 2020-09-05:
anyone can confirm the problem?
I also see the same wrong data into JobHistory list..

KenLucke, 2020-09-05:
I have seen this, too (different numbers)

Also, layer % completions of like 132413232%.  Unable to reliably reproduce, but it only seems to happen on large gccode files.

I'll try to grab a screenshot next time I see it.

KenLucke, 2020-09-05:
Another example:

[image]

Actual height of the model in Cura, BTW, is 50.594mm, and it's sitting directlly on the buildplate (no raft or anything else)

KenLucke, 2020-09-05:
[image]

AngusCox, 2020-09-06:
Same issue here!

[image]

spanzetta, 2020-09-06:
Thanks to all..

I wonder if the author (@StefanCohen) is reading these post..

KenLucke, 2020-09-06:
> I wonder if the author (@StefanCohen) is reading these post..

You know, most of these authors a) have other lives, b) don't make money doing this and have to feed themselves [and possibly others] in those other lives, and c) we don't know what they are going through in those other lives.

Give 'em a break.  It's only been 3 days since you opened this, and it's certainly a non-critical issue in a non-critical (albiet very useful) plugin.

KenLucke, 2020-09-06:
This is also a duplicate of #193 , for which there is a patch [#195] (I am just about to apply it, but it was confirmed that it worked by others)

spanzetta, 2020-09-07:
Thanks @KenLucke .. I didn't want to push (it was just a question) .. and I understand what you mean..

StefanCohen, 2020-09-07:
Thanks for the feedback. The Dashboard gets the current height from the DLP plugin. It looks like a recent DLP release may have increased the precision a bit and my rounding needs to be fixed as per the PR from Olli. I'll take a look at that and will close this ticket as a duplicate.

closed 2020-09-07 by StefanCohen

spanzetta, 2020-09-07:
I did the suggested manual fix and it works!
Thanks

AngusCox, 2020-09-07:
> I did the suggested manual fix and it works!
> Thanks

Same for me!
Thanks you all.