j7126/OctoPrint-Dashboard #199: Fan speed does not update when the speed is changed from the Marlin firmware screen

StarNamer, 2020-09-04:
**Is your feature request related to a problem? Please describe.**

Marlin firmware has an option REPORT_FAN_SPEED which causes the code to send a string "M106 P<fan> S<Speed>" on all serial ports,e.g "M106 P0 S127" for 50% on fan 0. With this enabled in firmware, it can been seen in the OctoPrint terminal window, but the fan gauge on the dashboard does not update.

**Describe the solution you'd like**

When the fan speed is changed and the report is sent from the Marlin firmware, the gauge should update to reflect the actual speed.

**Describe alternatives you've considered**

No obvious alternative.

**Additional context**

This may be a limitation in OctoPrint if the dashboard is simply reporting the current (last) speed set by OctoPrint, int which case I'll need to raise this as an issue there. However, since there doesn't see, to be a place where OctoPrint displays the fan speed, it's not clear whether it retains the value.

KenLucke, 2020-09-05:
I know it's notoriously hard to read settings back from the printer that are not saved in EEPROM, so this may not be possible for OctoPrint to do.  Not a programmer here, but I've seen the responses on other similar things that are done at the control panel not being able to be recognized back to OP.

StarNamer, 2020-09-07:
That's why I mentioned that Marlin has the option the send the string "M106 P<fan> S<speed>" to OctoPrint whenever the fan speed changes (whatever the cause). OctoPrint sees this, but, as far as I can tell, simply echoes it in the terminal window. Having looked further, OctoPrint-Dashboard reads the fanspeed variable from OctoPrint-DisplayLayerProgress, so it may be either that (or OctoPrint itself) which needs to process the string.

By the way, I am a programmer (still).

StefanCohen, 2020-09-07:
Thanks for the feedback. This is as you mentioned a limitation in OP. Getting the current fan speed reported from the printer F/W exposed thru the OP API is afaik not currently possible. This is likely due to the fact that fan speed reporting is not widely implemented in different firmwares (or even enabled since it is an optional config). I rely on DLP parsing of the gCodes sent to the printer to get the current fan speed so it is has its limitations but that is what is practically possible to do at the moment. That's the reason for closing this feature request.

closed 2020-09-07 by StefanCohen