j7126/OctoPrint-Dashboard #200: [Feature Request] - Print speed indicator

KenLucke, 2020-09-05:
I'd like to see [in among the static, non-addressable information displays, similar to the layer height and others] a simple indicator of the current print speed (in mm/second).  Should be able to be read from the G0/G1/G2/G3 Fxxxxxx gcode strings

StefanCohen, 2020-09-07:
Thanks for the feedback. I get the Idea but I don't see how practical it would be. A typical slicer will set different speeds for various movements. For example (My current print settings):

Infill: 100
Outer Wall: 50
Inner Wall: 60
Top Surface Skin: 50
Top/Bottom: 60
Travel: 250
Initial Layer Speed: 30

I imagine that that a current print speed would constantly jump between these values. During a typical layer it would switch like crazy between 250, 100, 60 and 50.

What do you think?

KenLucke, 2020-09-07:
Well, it's more of an OCD thing for me. :)  The layer height display indicator jumps constantly, too, as zhops and layer changes happen, so it would not be the only one jumping like crazy.  :)

 I guess it's the ex-pilot thing coming out in me, wanting to know every possible detail of what's going on on my "dashboard."

StefanCohen, 2020-09-07:
I understand. I'm not saying that it would be impossible or even undesired. I just don't see the value for me personally as it would be a blur on most of my prints. Maybe some kind of average speed during the last few seconds would be more interesting?

KenLucke, 2020-09-07:
> I understand. I'm not saying that it would be impossible or even undesired. I just don't see the value for me personally as it would be a blur on most of my prints. Maybe some kind of average speed during the last few seconds would be more interesting?

Maybe toggleable between the two?  Or Current/Average display (both), with the average time frame a preference?  I know, I'm just being OCD, as I said :)

j7126, 2020-09-27:
I think instead of a current display (as it would be constantly changing) what would be more practical would be Average(last 5 seconds)/Average(total)/Maximum

StefanCohen, 2020-09-28:
It is possible to get feedrate, feedrateG0 and feedrateG1 from DLP. The event handler in Dashboard already supports them but you will need to add the event listener for "DisplayLayerProgress_feedrateChanged". I would recommend to break it out to a separate event listner as that event would fire very frequently. Then aggregate the readings in an array and calculate an average using a rolling time window. The average can then be sent to the frontend less frequently together with the other stats.

StefanCohen, 2020-09-28:
Example: Add this to send_notifications in order to log the feedrates:

```
if event == "DisplayLayerProgress_feedrateChanged":
            self._logger.info("Feedrate: " + str(float(payload.get("feedrate")) / 60) + "mm/s")
```

The feedrates could then for example be added to a time-series data structure (list of dicts?) and the average could be calculated with pandas before it is sent to the front-end.

j7126, 2020-12-30:
ready for next release

ChuckRider, 2021-01-01:
Been running it through its paces this evening.  Looks great!  
Everything seems to be working as it should.  If I were going to provide 
any suggestions it would be to either remove the word Tool from the 
hotend temp gauge or add the words bed and fan to the other two.  That 
would provide a more consistent look.  That is being nit-picky though, 
it looks and works great.

Jefferey Neuffer wrote on 12/29/2020 10:55 PM:

closed 2021-01-11 by j7126

KenLucke, 2021-01-11:
Any thoughts on when this version will be released?

Willmac16, 2021-01-11:
The new version was just released. You may need to force refresh in the software update tab if you want it right now.

KenLucke, 2021-01-11:
Kewl.  I have a few prints going right now, when they're done, I'll install it. Thanks (in advance) for the new feature!