j7126/OctoPrint-Dashboard #201: Python 3 causes Pi CPU readings to dispaly Zero

MarkDerbyshire, 2020-09-16:
My SD card shat itself in my Pi 4.  Reloaded a new card with the image,  loaded all the plugins i use
Ran sudo apt update and sudo apt full-upgrade
Restored from a backup that was from a working SD Card (Internal temp was giving a reading)
Dashboard on this new installation now only shows the Pi 4 cpu temp as 0 deg.  I have disabled all plugins except Dashboard and DisplayLayer

All is up to date when running software update.

I installed Python 3 as a test and now all of the readings display as a Zero.

Pi 4
Octoprint Image - 1.17.0
Octoprint ver 1.4.2
Dashboard ver 1.14.0
Displaylayer ver 1.23.2

Installed in Argon1 Case with the fan script installed.  I also un installed the script but still showing zero

[image]

erichiller, 2020-09-16:
I can confirm the same here, after using the upgrade to Python3 script this happened, before that it worked fine.

Pi4 4GB, same versions as above.

Moskito99, 2020-09-16:
Same issue as #191

StefanCohen, 2020-09-20:
Closed as duplicate of #191

closed 2020-09-20 by StefanCohen