j7126/OctoPrint-Dashboard #202: Reduce dependency on DisplayLayerProgress

puckpuck, 2020-09-20:
**Is your feature request related to a problem? Please describe.**
With the layer shift problems in DisplayLayerProgress, people turning the plugin off, and the "won't fix" label on the [issue](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/124), it's time to consider and plan shedding dependencies on the plugin.

**Describe the solution you'd like**
Though some widgets like Layer Durations may not be trivial, other widgets like Fan Speed, could be implemented with less effort. Doing this will allow people to continue to use this plugin with all functionality enabled, and not be stressed by another plugin that could potentially cause issues in your prints.

StefanCohen, 2020-09-20:
Thanks for the feedback. Are you suggesting that I remove the widgets that depend on DLP?

puckpuck, 2020-09-21:
I'm suggesting to keep the widgets but change how the widgets get data.

trollhunterD, 2020-09-22:
I 2nd this, I enjoy your plugin but I  just can not use displaylayerprogress with my printers since it causes layers shifting on all my printers.

StefanCohen, 2020-09-22:
Thanks. I will most likely not try to re-implement the functionality provided by DLP myself. That task is simply too daunting for me and and was the reason I went with the DLP dependency in the first place.

trollhunterD, 2020-09-22:
I just found out you added the ability to suppress the install DLP  message.

StefanCohen, 2020-09-27:
It doesn't look like anybody has identified the cause for the layer shifts when using DLP (on prusa printers?). The problem  seems intermittent and very vague to me when reviewing the ticket. I think identifying and fixing this issue in DLP is the top priority. The alternative would be to implement metric f*ckton of regex GCode parsing in the Dashboard as a replacement and it might very well end up causing the samt symptoms as with DLP :)

Willmac16, 2020-09-30:
As a new maintainer of this plugin and a Prusa owner who has noticed some DLP layer shift symptoms, I will be looking into this.

Willmac16, 2020-10-08:
I helped test some new changes to DLP that appear to have fixed the problem for me. There should be a new release out now.

Willmac16, 2020-10-09:
@puckpuck let me know if the new release of DLP has fixed your shifting problems.

puckpuck, 2020-10-09:
I just re-read all the activity in the past 10 days on the issue and will take a look at it over this weekend.

puckpuck, 2020-10-13:
Haven't tried a print specifically that failed before, but I did run a longer print yesterday with gyroid infill and no issues at all.  I'm going to close this, assuming DLP's M117 throttle fixed the problem.

closed 2020-10-13 by puckpuck