j7126/OctoPrint-Dashboard #203: Looking for a dashboard maintainer

StefanCohen, 2020-09-21:
I no longer have the time to maintain or enhance this plugin due to other commitments so I'm looking for someone to take over this plugin. Let me know if you are interested.

Regards,
Stefan

gitzk2, 2020-09-22:
Too bad, I really like this plugin.  I don't think I am qualified to take it over as I am just starting to learn python.  If I was I would not mind.  Just out of curiosity what does it take to maintain or even create a plugin?  Hopefully someone will come along to take it over though.

StefanCohen, 2020-09-22:
Plugin development is split between Python for the backend and JS for the frontend. There is an excellent [tutorial](https://docs.octoprint.org/en/master/plugins/gettingstarted.html) in the OP documentation that I used when I started. I had very limited coding experience when I started (as clearly shown in the code) but I got som very good support from the community in the OP forum. You can follow the progress from the beginning in this https://community.octoprint.org/t/dashboard-plugin/11536/24.

Willmac16, 2020-09-22:
I would be willing to take over, but if a more zealous soul should appear, I also would be happy to leave it to them.

j7126, 2020-09-23:
I also would be open to taking over the plugin.

StefanCohen, 2020-09-24:
@Willmac16 @j7126 It looks like you both are interested in taking over maintenance and possibly do it together. Is that correct? If you are up for it, then maybe we should have a call to discuss the forms for the transition? I don't know if that is possible but I'm in CET+1 and can set up a hangout if we can find a timeslot that works for everybody. What do you think?

Willmac16, 2020-09-24:
I'm a considerable number of time zones away(UTC-7), and my day is starting shortly, so I'm unsure if I could schedule a call before your tomorrow.

StefanCohen, 2020-09-24:
Ah. Maybe we can do it during the weekend? I can walk you thru the history, code, dependencies, 3:d-party libraries, some gotchas, the release process, etc that might be useful to you. I can also show you how I have set up my dev env if that would be interesting. We could even try to solve a bug or two and push a release together so you get a feeling for how I have been doing it so far. We can discuss what to do with the semi-ready release that I have here locally that I worked on back in June.
And then there is the matter of how to transfer ownership.

Let's see what @j7126 says

Willmac16, 2020-09-24:
Sounds good.

j7126, 2020-09-24:
Ok, 
Sounds good
My time zone is UTC+10

j7126, 2020-09-24:
@Willmac16 @StefanCohen may I suggest that we continue this conversation elsewhere than this issue. 
I would suggest discord.

StefanCohen, 2020-09-24:
Cool. I'll set up a discord for the conversation.

StefanCohen, 2020-09-24:
@Willmac16 Pleas send your discord ID via email and I'll invite you too.

morphias2004, 2020-09-25:
> 
> 
> I no longer have the time to maintain or enhance this plugin due to other commitments so I'm looking for someone to take over this plugin. Let me know if you are interested.
> 
> Regards,
> Stefan

Noooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooooo.........   😭 

I love this plugin and have it running on my TV so others can see my prints when they're at my house.

@Willmac16 and @j7126 - you are legends if you continue @StefanCohen's awesome work!  👍

j7126, 2020-09-27:
@Willmac16 and I have taken over maintenance of the plugin.

closed 2020-09-27 by j7126