j7126/OctoPrint-Dashboard #205: Add filament change indicator to "Extruded Filament" widget

StefanCohen, 2020-09-27:
**Is your feature request related to a problem? Please describe.**
I rarely change filament during a print but It would be nice to get an ETA on the next filament change when I do.

**Describe the solution you'd like**
DLP provides a metric called "changeFilamentTimeLeftInSeconds" and "estimatedChangedFilamentTime" that would be a nice addition to the "Extruded Filament" widget.

Example:
Extruded Filament: 2.7m
Next Filament Change: 00:01:14

**Describe alternatives you've considered**
None

**Additional context**
Add any other context or screenshots about the feature request here.

j7126, 2020-10-15:
It seems the plugin already monitors changeFilamentTimeLeftInSeconds. so all that is left to do it actually add the widget.

StefanCohen, 2020-10-15:
Correct. I think it could work to add the metric to the existing filament widget as a secondary field to make it dual purpose and increase the density rather than as a separate widget. Something like this:

[image]

j7126, 2020-10-15:
I was planning to add it as a separate widget because there is just empty space there anyway and I think it would be more clear what it is.

j7126, 2020-10-16:
ready for next release

closed 2020-10-16 by j7126

ruedli, 2021-02-26:
I can see the indicator, but it just shows a "dash" (-). I was hoping to see it predict when it changes filament for the MMU2. This implies GCODE T0, T1, T2, T3 and T4. Is it indeed triggered by these GCODEs? Or is it with M600 gommands?

Willmac16, 2021-02-27:
@ruedli previously this statistic was coming from display layer progress and tracked the time until the next `M600`. At this point, Dashboard is moving away from depending on DLP due to some issues found while running it. Right now, I am working on reimplementing the time to filament change and will aim to support both `T#` changes and `M600` changes.

Willmac16, 2021-02-27:
@ruedli if you would like to try the new version it is available to install in the plugin manager with the url `https://github.com/j7126/OctoPrint-Dashboard/archive/dlpreplacement.zip`.

ruedli, 2021-02-28:
@Willmac16 OK, I will try! Currently I have an MMU2 job active with 4.5
hours to go. When you are processing the GCODE yourself, I have a candidate
for a useful information. Often I have MMU printjobs, where after a certain
time, no color changes are necessary anymore. Then they typically continue
reliably, without need for user intervention. Knowing this time is useful
to me. So something like: "Last colorchange in: H:MM".

ruedli, 2021-03-02:
I installed via URL, but did not see a huge difference, am I supposed to
uninstall the regular plugin first?

[image: image.png]

Cheers Ruud

Willmac16, 2021-03-02:
@ruedli You will need to reupload files so that the dashboard backend can process them.

ruedli, 2021-03-05:
True,

It now showed the indicator, but the values are not OK.

It starts with an estimated of 18.5 minutes, hoever: the first layer is
1:14:00 (so 75 minutes!) and the filament change is in the second layer
(takes 19 minutes) towards the end of it,

At the moment is changes layer, it still thinks it is only 10 minutes.

Further the amount of extruded filament seems off: it did not refresh, but
after I pressed F5 it showed 126.39 meter! The estimations for print time
seem about twice as much as I hope they will be... If you want, I can send
you my GCODE, but you need to tell me how I can send you a big file.

[image: image.png]

Thanks for your help,

Kind regards,
Ruud

Willmac16, 2021-03-05:
My current solution to estimating time to next change was based off of number of move commands and the total time estimate octoprint had. If you don't have PrintTimeGenius installed that estimate will be more off, but I should probably look into better processing to get more accurate time estimates.

ruedli, 2021-03-05:
I have printtimegenius installed (at least I had it in my python2 setup). I
installed everything from scratch and restored my settings and plugins
using the latest octopi so with python3.

I cannot see for usure what I have installed, as I have the 14h print
running and now there is a (but it is stil printing):

[image: image.png]

The Pi is not overloaded, it is an Pi4, running from SSD and uses very few
resources.

It happened as I moved to the dashboard tab, but it could very well be
because of other plugins, as also Octolapse had an "internal error".
Would you recommend to go to the main branch for octodash? I noticed it is
also sending M117 commands at each layer change, but maybe that is
intentional.

Cheers Ruud

Willmac16, 2021-03-05:
@ruedli it doesn't seem like the images you are attaching from email are coming through.

ruedli, 2021-03-05:
Sorry for the confusion, I updated them, added the images (.png files are linked).

Note that my "first layer" is a square of 180x180 mm, so at low speed, those diagonals take a long time (75 minutes was not so far off). The second layer prints faster (19 minutes), but has more move commands, because there are 36 characters in a different color in them. This might "throw off" your calculations, The Filament change is towards the end of the second layer (about 5 minutes before the end. From start to 1st tool change it takes about 90 minutes.