j7126/OctoPrint-Dashboard #211: Add print time graph

StefanCohen, 2020-10-01:
The gauges shown during a print job includes "Job" and "Layer". The current Job gauge is really displaying the bytes sent vs the size of the file size. 

It would be nice to rename the current "Job" gauge to "Gcode" and add a third gauge named "Time" that displays the time spent vs the estimated time in %. The reason is that the Gcode gauge is rarely linear while the proposed Time gauge should be (at least with PTG installed). A third gauge would also result in a more symmetrical UI for most configurations and it would be easier to know the time progress at a glance compared to reading the value down in the ETA widget

Something like this:

[image]

An equivalent bar chart needs to be added of course.

j7126, 2020-10-11:
Ready for next release

StefanCohen, 2020-10-13:
The Time Gauge sometimes (but not always) shows the percentage as NaN when starting a print. I guess the timestamps needs to be initialised to some sane values on print start.

j7126, 2020-10-13:
I thought I already fixed that
are you running the latest dev commit?

closed 2020-10-15 by j7126