j7126/OctoPrint-Dashboard #212: Accommodate loooong filenames

xeddog, 2020-10-04:
**Describe the solution you'd like**
When the dashboard is displayed, long filenames continue outside of the box. As shown in the attached sreenshot.  

[image]

Perhaps consider reducing the fontsize on long filenames, and/or wrap to two lines.

StefanCohen, 2020-10-04:
The Dashboard already supports reducing the fontsize on long filenames but It looks like minSize:5 in the following code in dashboard.js is a bit too big fontsize for resizing some really long filenames properly. Changing it to 2 will accommodate at least 80 chars.

`self.fitties = fitty('#fileInfo', { minSize: 5, maxSize: 20 });`

It is of course not viable to keep scaling down the font like this forever so some method to either wrap or truncate would be preferred.

closed 2020-10-05 by Willmac16

KenLucke, 2020-10-08:
LOL, someone was complaining about LONG filenames?

Accommodate this :)

![Uploading CR-10 v2 (Yellow) LAN [OctoPrint] - 0% 2020-10-08 14-40-07.jpg…]()