j7126/OctoPrint-Dashboard #214: [BUG] Octoprint refuses to restart or reboot with Dashboard v1.15.0

fnsign, 2020-10-05:
**Describe the bug**
The latest release causes problems when restarting or rebooting Octoprint. A restart dialogue appears though, but after confirming the restart nothing happens any more. The UI is obviously still operational, but Octoprint is not restarting, but throus an exception (see attached log). Even restarting Octoprint from ssh cli is not helping. A reboot is necessary, which even takes a few minutes.
Rolled back to 1.13 and the issue is not appearing any more

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Restart Octoprint' or "Reboot and confirm the prompt. 

**Expected behavior**
after confirming the prompt Octoprint should restart and display the known message to "Attempt to re-connect" and the "Reload". After restart the UI should become available again.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log
[octoprint-4.log](https://github.com/j7126/OctoPrint-Dashboard/files/5329826/octoprint-4.log)

**OctoPrint server and Plugins:**
 - Platform: RPi 3b+
 - OctoPrint version: OctoPrint 1.4.2
 - OctoPi version: OctoPi 0.17.0
 - Dashboard version Dashboard 1.15
 - DisplayLayerProgress version [e.g. 1.14.0]

**Desktop (please complete the following information):**
 - OS and version: macOS Catalina 10.15.6
 - Browser and version Safari 14.0

j7126, 2020-10-06:
I cannot seem to reproduce your issue. You have quite a few plugins installed there and I tested it with all of those plugins installed on my OctoPrint instance as well as the latest dashboard and the issue is not appearing (however I cannot know exactly what settings you have set in those plugins or OctoPrint). Looking at the log I can't see any errors that seem like they would be specifically related to dashboard. Given that you can solve the issue by rolling back to an earlier version of dashboard it seems like there may be some conflict between the latest version of dashboard and one of your plugins. 

If you could install the latest dashboard and disable each of the other plugins you have installed one by one to see which one may be causing the conflict, that could help us identify the issue.

fnsign, 2020-10-06:
Thanks for your prompt response!
Your assumption was right that the issue is caused due to intereference of the Dashboard v1.15.0 plugin with another.
I deactivated all 3rd party plugins and re-activated them 1-by-1. The problem occurs with this plugin: [WS281x _LED_Status](https://github.com/cp2004/OctoPrint-WS281x_LED_Status) Maybe you could have a look on it why that happens? With Dasboard 1.13.0 there is no issue at all.

cp2004, 2020-10-06:
@j7126 - hey there, author of WS281x LED status here. 

Common cause of this problem is using non-daemon threads in Python + multiprocessing + OctoPrint's shutdown handler don't play nice. Hence the binding never gets released, so you get the 'address already in use error'

I'll take a look later, but if you want to look before I get a chance, that's where I would start.

cp2004, 2020-10-06:
This is 100% it - you use `octoprint.util.ResettableTimer` and by default it is not set to `daemon=True`. Pass it as a kwarg to the class.

I'll send in a PR later if this is not clear, I'm on my phone at the moment 🙂

cp2004, 2020-10-06:
PR created, but not tested. Will give it a go later. 🙂

cp2004, 2020-10-06:
Confirmed #219 is the fix. And filed PR https://github.com/OctoPrint/OctoPrint/pull/3760 to make sure this doesn't happen again, from 1.5.0 release.

Willmac16, 2020-10-06:
Thank you for the PR. The commit should go out with release 1.15.1.

StefanCohen, 2020-10-06:
New PR submitted for review.

j7126, 2020-10-06:
merged

closed 2020-10-06 by j7126