j7126/OctoPrint-Dashboard #216: Command widget enhancements

StefanCohen, 2020-10-06:
Possible enhancements for the command widgets:

1. Better error handling: Check the return code of the called command and log an error including whatever is in stderr. Handle long-running commands gracefully by terminating them before they are called again and log the reason.
2. Add a modal setting dialog for editing commands. This would allow viewing longer command strings without scrolling and:
3. Add a test button to the UI that allows the user to test the call before saving it. display the result in a modal dialog.
4. Add an enable/disable setting to each command. Disabled commands will not be executed nor displayed in the Dashboard.
5. Add a setting for the polling interval. The hardcoded 60s can be way too long or short depending on the command called.
6. Truncate the displayed command result in the widget to fit the size of the widget container. (text-overflow: ellipsis)?
7. Possibly add a second command for each command widget. This would make it possible to display two metrics in one widget similar to how it is done in the ETA widget. This would provide a higher metric density compared to adding two widgets (ex: temp & humidity sensor). The CPU/SOC widget currently holds three metrics but that is probably overkill.

Willmac16, 2020-11-06:
As of this post, the following should be done: 2, 3, 4, and 5.

ShagoY, 2020-11-22:
Hello,

It is planned to add simple GCODE command ?

Willmac16, 2020-11-22:
Do you mean buttons to run Goode on the printer? If so, see issue #232 that will likely be added in v2.

ShagoY, 2020-11-23:
I had seen this outcome but as you don't mention the term GCODE, I thought it was more an integration than anything else.

But yes, I'm talking about being able to create commands that run GCODE and not just scripts :)

In my case, it's to integrate commands from the [WS281x_LED_Status plugin](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/wiki/Features), for example activate the torch or change the colors of my leds etc...

Thanks in advance !