j7126/OctoPrint-Dashboard #217: Webcam enhancements

StefanCohen, 2020-10-06:
Possible webcam enhancements:

1. Click image to zoom. This will display the image (scaled proportionately) in a modal view that fills the viewport. Click image again to get back.
2. In unzoomed mode, set opacity with a transition on hover for the image to give a UX hint that it is possible to click the image. Like this example: https://www.w3schools.com/howto/howto_css_modal_images.asp
3. Add setting in the webcam settings (available if multicam is enabled) to enable cycling thru the streams. Add setting for cycle time (default 5s?). Add button next to enable/disable webcam in Dashboard screen to pause/resume cycling

j7126, 2020-10-06:
Also picture in picture

StefanCohen, 2020-10-06:
Maybe add a timeout to each camera config (default 5s?). Show the loading animation until the stream is fetched and a broken stream icon if the timeout is reached.

StefanCohen, 2020-10-06:
Also: a button in the camera settings config modal to test the stream, similar to how OP does it.

j7126, 2021-01-31:
Additionally: when the full screen and full browser buttons are disabled the toggle webcam button looks a bit lonely up there, it should probably be moved to the top right corner of the webcam. 
[image]