j7126/OctoPrint-Dashboard #218: detect missing GCODE Viewer viewModel

StefanCohen, 2020-10-06:
Whenever a user disables the GCODE Viewer, the Dashboard Viewmodel fails to bind and the frontend crashes halfway thru rendering. This is the most frequently reported issue and users have no way of detecting the error unless they open the browser console.

Solution: detect the issue before binding, log the error and notify the user via the ui (pNotify). Optionaly, also disable the layer gauge and data binding until the GCODE viewer has been enabled to allow the rest of the dashboard to work.

See here for an example of a ticket: https://github.com/j7126/OctoPrint-Dashboard/issues/171

Willmac16, 2020-10-17:
Additionally, it would be nice to have behavior to detect if a GCode file is not being automatically visualized due its size.

Willmac16, 2020-10-31:
I'm going to close this issue because I think it would be better to work on getting the data from another source than to work on improving this method.

closed 2020-10-31 by Willmac16