j7126/OctoPrint-Dashboard #224: Is 1.15.1 incompatible w/ Octoprint 1.3.12?

ProfEngr, 2020-10-15:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
I get this error in the logs:

2020-10-15 17:25:05,929 - octoprint.plugin.core - ERROR - Error loading plugin dashboard
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 10, in <module>
    from octoprint.access import ADMIN_GROUP
ImportError: No module named access

**To Reproduce**
Steps to reproduce the behavior:
Updated from 1.10.x to latest via software update.

**OctoPrint server and Plugins:**
 - Platform: RPi 3b+
 - OS and version: Raspian
 - OctoPrint version: 1.3.12
 - OctoPi version: 0.17.0
 - Dashboard version: 1.15.1
 - DisplayLayerProgress version: 1.24.0

**Desktop (please complete the following information):**
 - OS and version: Windows 7
 - Browser and version: Firefox 81.0.1

StefanCohen, 2020-10-15:
It looks like the permission request for the cmd widgets is incompatible with earlier versions of OP. Maybe do a version check before requesting permissions could solve this?

j7126, 2020-10-15:
I will look into this
it may be a bit more complex than that because there are a few places where it tires to check for permissions

ProfEngr, 2020-10-15:
OK, I rolled back to 1.13.0 for now.

j7126, 2020-10-15:
I believe you can still run octoprint 1.4.2 on python 2

j7126, 2020-10-15:
@ProfEngr once I have made the fix in a few minutes I will give you a different link to try to install the plugin from so you can test if it works

j7126, 2020-10-15:
@ProfEngr please try install dashboard via the plugin manager using this link:
https://github.com/j7126/OctoPrint-Dashboard/archive/%23224.zip

ProfEngr, 2020-10-15:
Different error this time it seems.
2020-10-15 19:03:16,863 - octoprint.plugin.core - ERROR - Error loading plugin dashboard
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 15, in <module>
    from octoprint.access.permissions import Permissions
ImportError: No module named access.permissions

It made no difference whether I disabled multicam or not.

j7126, 2020-10-15:
@ProfEngr I have pushed a new commit to hopefully fix that. can you try reinstalling again from that link (you may have to clear cache)

ProfEngr, 2020-10-15:
Fix one, uncover another?

2020-10-15 19:23:56,544 - octoprint.plugin.core - ERROR - Error loading plugin dashboard
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 863, in _import_plugin
    instance = imp.load_module(module_name, f, filename, description)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 13
    from octoprint.access.permissions import Permissions
    ^
IndentationError: unexpected indent

j7126, 2020-10-15:
@ProfEngr try now

ProfEngr, 2020-10-15:
That loaded in the UI. I got a unicode warning from 

2020-10-15 19:36:36,672 - py.warnings - WARNING - /home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py:658: UnicodeWarning: Unicode unequal comparison failed to convert both arguments to Unicode - interpreting them as being unequal
  if value_source != value_target:

I don't know what it's related to though.

Thanks for the speedy updates!! Now to get both cams on the Control Tab. Off to the settings I go....

j7126, 2020-10-15:
@ProfEngr I have pushed the fix in the 1.15.2 release. you should be able to reinstall the plugin now normally from the plugin repository using the plugin manager. make sure you do this otherwise you will not get updates in the future.

closed 2020-10-15 by j7126

cp2004, 2020-10-19:
> I still have a few plugins I use that are reported to not be updated for Python3 so I'm stuck on the older Octoprint version.
Just to clarify: 1.3.x is more than a year old (in terms of development.

See this for the explanation: https://community.octoprint.org/t/does-octoprint-1-4-0-and-later-only-run-under-python-3-now-did-octoprint-stop-supporting-python-2/18322?u=charlie_powell

commit 6d8de561 (2020-10-15): try again to fix #224
commit 0b002470 (2020-10-15): #224 fix errors on octoprint 1.3 (#226)

* fix for #224

* fix for #224

* try again to fix #224

* fix indentation error #224