j7126/OctoPrint-Dashboard #227: Can't switch between 2 cameras

ProfEngr, 2020-10-15:
Thanks again for the quick resolution for the older Octroprint version.

I got into the settings and set up both of my USB cameras using the URLs from the now-defunct Multicam plugin. I can still only see the default cam in the Control tab (no button to switch between the 2), but when I add the webcam widget to the main tab I get the second stream in an overly-large window for its puny 640x480 res.

Did I miss something obvious?

j7126, 2020-10-15:
@ProfEngr can you please post a screenshot of the dashboard tab and of the dashboard webcam settings so I can see exactly what you mean

ProfEngr, 2020-10-16:
OK, I missed the icons for switching on the main tab. The nozzle is still huge for its native resolution though. The other cam is a 1280 so it shows up ok. There are no camera-like icons on the Control tab to swap streams. I couldn't get the entire UI in the shot. Hope these are ok.
Control Tab, no selector icons anywhere that I saw
[image]
Main Tab, has selector icons
[image]

ProfEngr, 2020-10-16:
Oh, settings...
[image]

j7126, 2020-10-16:
The dashboard multi-cam implementation only shows multiple cameras in the dashboard and does not show them in the control tab. 
You will still need to use the multicam plugin if you want multiple cameras in the control tab. 
Regarding the size of the image for your lower resolution camera, you can try playing around with aspect ratio settings to fix this.

ProfEngr, 2020-10-16:
I'm using the 4:3 for the low res camera.

Wait, I thought the changelog said multicam wasn't compatible anymore. Was that just meaning the Main tab? If it still works w/ Control then we're good as I had that working before.

j7126, 2020-10-16:
> Was that just meaning the Main tab? 

yes

closed 2020-10-16 by j7126