j7126/OctoPrint-Dashboard #229: gauges don't adjust when fan gauge off

TomW1605, 2020-10-20:
**Describe the bug**
when the fan gauge is turned off the other gauges are not respaced or sized to fill the gap. this makes it look like there is a big hole there.

**To Reproduce**
1. turn off the fan gauge in settings
2. look at dashboard

**Expected behaviour**
the other gauges on that row should move so that they are still evenly distributed across the space

**Screenshots**
[image]

**OctoPrint server and Plugins:**
 - Platform: RPi 2b
 - OS and version: Raspbian Stretch
 - OctoPrint version: OctoPrint 1.4.2
 - OctoPi version: Octopi 0.16.0
 - Python version: Python 2.7.13/3.5.3
 - Dashboard version: 1.15.2
 - DisplayLayerProgress version: 1.24.0

**Desktop (please complete the following information):**
 - OS and version: Windows 10
 - Browser and version: chrome 86.0.4240.75

**Additional context**
Add any other context about the problem here.

StefanCohen, 2020-10-20:
Just keep in mind that there may be multiple hotend gauges as well as a heated chamber gauge in there as well (depending on the printer profile).

j7126, 2020-10-20:
I will do this tommorow

j7126, 2020-10-21:
Will be in next release

closed 2020-10-21 by j7126