j7126/OctoPrint-Dashboard #232: Enclosure Plugin combination

jlogan223, 2020-11-06:
I love this dashboard plugin, and I use the encolsure plugin a lot to, but I am always having to flick back and forth between the tables. Is it possible to incorporate the enclosure plug in with the dashboard. For example I have switches to operate lights, printer power, fans etc, and have actions thag operate based on Temperatures. As well as have certain GPIO'sactivate when a print is complete. Could we get these buttons added to the dashboard plugin?

StefanCohen, 2020-11-08:
I think you are asking for a way to incorporate controls in the dashboard that can perform actions in the enclosure plugin. Is that correct? @Willmac16 @j7126: Something to consider for v2?

j7126, 2020-11-08:
> Something to consider for v2?
I think so

jlogan223, 2020-11-11:
> I think you are asking for a way to incorporate controls in the dashboard that can perform actions in the enclosure plugin. Is that correct? @Willmac16 @j7126: Something to consider for v2?

Yes that would be amazing. :-) If we could even drag and drop the arrangement of the Dashboard tab would be a bonus as well. 

You guys and Ladies are doing some awesome work.

j7126, 2021-01-11:
This is definitely something that will be possible with v2 of dashboard. I am closing this issue for now, as we are moving all discussion related to v2 to #249

closed 2021-01-11 by j7126