j7126/OctoPrint-Dashboard #233: Webcam flipV enabled only if flipH is enabled too

lucabartoli, 2020-11-09:
**Describe the bug**
I'm trying to flip the webcam vertically only but seems impossible.
Vertical flip is enabled only if Horizontal flip is enabled too.
Horizontal flip works alone as expected.
Vertical flip doesn't work alone.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to the plugin's webcam settings
2. Click on edit
3. Enable vertical flip only
4. click done and save
5. The image is not vertically flipped
6. go back to the webcam settings
7. Vertical flip is disabled 

Try again enabling horizontal and vertical flip, it works.

**Expected behavior**
Webcam can be flipped vertically only.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log
* Browser console log
* gcode file that can be used to reproduce the problem

**OctoPrint server and Plugins:**
 - Platform: RPi 3b+
 - OS and version: Raspbian 10
 - OctoPrint version: 1.4.2
 - OctoPi version: 0.17.0
 - Python version Python 2.7
 - Dashboard version 1.15.2
 - DisplayLayerProgress version 1.24.0

**Desktop (please complete the following information):**
 - OS and version: Win 10
 - Browser and version Chrome latest

j7126, 2020-11-09:
Hi @lucabartoli 

I have tested this on my octopi instance and can't seem to reproduce this behavior. 

Can you please try restarting your OctoPrint server and clearing your browser cache and cookies, and see if that helps the problem. 

Can you also try disabling dashboard multi camera and set vertical flip in the default octoprint webcam, to see if that flip gets shown by dashboard.

j7126, 2020-11-09:
Can you please also include your logs

lucabartoli, 2020-11-09:
I tried clearing the cache and restarting octoprint, but didn't solve the issue.

These are the screenshots I forgot to upload before:
Setting webcam flip
[image]

The webcam is still reversed
[image]

Going back to settings, check mark disappeared
[image]

I updated to python 3 in the meantime, trying to solve the issue.
Now the webcam works, but I can't use the "toggle webcam" button. If I press it, the webcam disappears (as expected) and toggle webcam button doesn't work anymore (seems not having effect).
Reloading the page always works to make the webcam coming back.

j7126, 2020-11-09:
Can you please upload octoprint's log file and also check if there are any errors in the browser console.

lucabartoli, 2020-11-09:
[octoprint.log](https://github.com/j7126/OctoPrint-Dashboard/files/5510772/octoprint.log)

And this is the console
[image]

Willmac16, 2020-11-09:
Do you have Dashboard multiple camera support enabled?
[image]

Willmac16, 2020-11-09:
It looks like I fixed the original bug you reported in 1.16, and j7126 fixed the toggle webcam button bug.

j7126, 2020-11-10:
1.16.2 released

closed 2020-11-10 by j7126

lucabartoli, 2020-11-10:
My multiple camera support was disabled, but the update fixed the button!
Kudo to you for the great and instant support! I really appreciate your effort