j7126/OctoPrint-Dashboard #234: DisplayLayerProgress Error

pingywon, 2020-11-09:
Wokr fine until the latest update. 

Now On my dash I get a Warning! Can't get stats from DisplayLayerProgress. Is it installed?

It is indeed installed.

OctoPi 0.17.0
Octoprint 1.4.2 (all patched and updated)
Python 2.7.16

Willmac16, 2020-11-09:
Could you please send Octoprint log files from the raspberry pi and a screenshot from the browser console.

pingywon, 2020-11-10:
[image]

I cleared the logs and restarted the whole system.

the plugin_DisplayLayerProgress_events.log is empty.

The only relevant line for dashboard is:

2020-11-10 07:32:12,104 - octoprint.plugins.dashboard - INFO - Dashboard started

No other mention and DisplayLayerProgress starts up fine in the log also.

StefanCohen, 2020-11-10:
That is a symptom of a JS error in the browser. Please check your browser console for errors and copy the result here as @Willmac16 suggested.

pingywon, 2020-11-10:
My bad. I didn't catch that ask...

here:

`Could not bind view model DashboardViewModel to target #tab_plugin_dashboard : ReferenceError: Unable to process binding "foreach: function(){return commandWidgetArray() }"
Message: Unable to process binding "visible: function(){return (enabled && enabled()) &&($parent.settingsViewModel.settings.plugins.dashboard.showCommandWidgets() && $parent.fsCommandWidgets()) }"
Message: enabled is not defined
    at visible (eval at parseBindingsString (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:6802), <anonymous>:3:60)
    at update (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:7:9356)
    at a.B.i (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:9069)
    at Function.Pc (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:16228)
    at Function.Qc (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:16006)
    at Function.aa (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:15847)
    at Object.a.m.a.B (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:14897)
    at http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:9056
    at Object.q (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:1:145490)
    at m (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:8580)
    at k (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7607)
    at Object.a.Rb (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:4:1427)
    at http://192.168.0.116/static/webassets/packed_libs.js?df208d52:7:13045
    at b (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:7:12697)
    at c (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:7:12993)
    at p (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:7:14379)
    at a.a.Bb (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:9:1679)
    at Object.w (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:10149)
    at a.B.i (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:7:14730)
    at Function.Pc (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:16228)
    at Function.Qc (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:16006)
    at Function.aa (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:15847)
    at Object.a.m.a.B (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:14897)
    at Object.a.kd (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:7:14568)
    at Object.update (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:8:59)
    at update (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:7:3098)
    at a.B.i (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:9069)
    at Function.Pc (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:16228)
    at Function.Qc (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:16006)
    at Function.aa (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:15847)
    at Object.a.m.a.B (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:2:14897)
    at http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:9056
    at Object.q (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:1:145490)
    at m (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:8580)
    at k (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7607)
    at g (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7660)
    at g (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7660)
    at g (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7660)
    at g (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7660)
    at Object.a.Rb (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:4:1427)
    at http://192.168.0.116/static/webassets/packed_core.js?3fb354c1:878:8
    at Pn (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:1:92204)
    at Function.<anonymous> (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:1:102635)
    at http://192.168.0.116/static/webassets/packed_core.js?3fb354c1:873:26
    at Pn (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:1:92204)
    at Function.<anonymous> (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:1:102635)
(anonymous) @ packed_core.js?3fb354c1:879
Pn @ packed_libs.js?df208d52:1
(anonymous) @ packed_libs.js?df208d52:1
(anonymous) @ packed_core.js?3fb354c1:873
Pn @ packed_libs.js?df208d52:1
(anonymous) @ packed_libs.js?df208d52:1
bindViewModels @ packed_core.js?3fb354c1:866
sentryWrapped @ packed_core.js?3fb354c1:430
setTimeout (async)
(anonymous) @ packed_core.js?3fb354c1:430
(anonymous) @ packed_core.js?3fb354c1:883
u @ packed_libs.js?df208d52:1
fireWith @ packed_libs.js?df208d52:1
[... 87 lines cut ...]
    at Object.update (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:8:59)
    at update (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:7:3098)
[... cut ...]
    at k (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7660)
    at g (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7660)
    at g (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7660)
    at g (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7498)
    at k (http://192.168.0.116/static/webassets/packed_libs.js?df208d52:3:7660)`

StefanCohen, 2020-11-10:
It looks like that problem was addressed in 1.16.1. Please upgrade to see if that resolves the issue.

StefanCohen, 2020-11-10:
@Willmac16 would it be possible to catch all binding errors and display an error message to make it more obvious where to look for the problem?

Willmac16, 2020-11-10:
I was able to recreate the issue on my end by editing the config.yaml file. If you just want your instance to work while we work on patching the issue you can edit the config.yaml file on the raspberry pi so that the command widget section appear like this `
commandWidgetArray:
        -   command: echo "Test CMD Widget"
            enabled: false
            icon: command-icon.png
            interval: '10'
            name: Default`

The enabled line is all that needs to be added.

closed 2020-11-11 by pingywon