j7126/OctoPrint-Dashboard #235: Restrict Webcam via Access Control

charredchar, 2020-11-11:
Honestly I am not sure if this should be a bug or feature request... But I don't think it was taken into account to start with so it isn't a bug more than it should be a feature to be implemented.

OctoPrint allows users to be restricted to specific things via the Access Control settings, one of these settings is the Webcam feed. While the webcam feed on the Control Tab does indeed get blocked for this user the webcam feed on the Dashboard tab still streams video from the camera and it should not.

While my specific use case is that I am attempting to run a Kiosk user on my Pi and a webcam stream causes issues, this is really something that should be looked at because of access restrictions people might want on specific users. In my case I want to be able to access the webcam with my admin account from my desktop or cell phone but I don't need it for the Kiosk user.

Willmac16, 2020-11-11:
First, for the kiosk use case you should be able to disable webcam viewing in full screen mode (assuming you have the latest version of Dashboard) while still retaining the webcam view in the tab view.
Second, webcam access control definitely is something that we could implement, with the major caveat that while using the default octopi setup, the webcam can be directly accessed regardless of access control settings.

charredchar, 2020-11-11:
I'm not using fullscreen for my Kiosk user, but it's more that I don't want the user to see the stream at all so I'd like it to be disabled in the tab view as well. Since OctoPrint is going to start requiring Access Control anyway thought it might be a good idea to just go along with that setting. They already have a restriction that stops the stream in the control tab.
I have heard about the second part before. Since OctoPrint relies on an external service for the stream (mjpg-streamer) you can't really block access completely within OctoPrint. But access restrictions is a start and helps just stop anyone who is logged in from seeing it easily.

StefanCohen, 2020-11-11:
That leads to the next question. Should there be access control on individual widgets regardless of type?

charredchar, 2020-11-11:
This did come to mind when I was typing out the part of OctoPrint requiring Access Control in the future. The "easiest" would probably be just the items in Dashboard that have a direct link to an item in Access Control. An example of this would be something like the Print/Pause/Cancel commands, though I have not personally tested if these gets effected by Access Control already.

closed 2021-01-11 by j7126