j7126/OctoPrint-Dashboard #236: Several issues since update

jlogan223, 2020-11-12:
Hello, not sure if you are aware of these issues, but since updating to the latest version yesterday I seem to have a few issues:
[image]

None of the .gcode data is being pulled in, Command widget image is broken (Do we have any instructions on how to set up the command widgets? and an example of some code, such as toggling a GPIO state?) The Camera no longer shows and the LayerProgressPlugin is installed and use to work fine, but now it no longer sees it.

Lefuneste83, 2020-11-12:
Same issue here on freshly installed Raspi 4. Reinstalled DisplayLayerProgress Plugin without any change to the error message above.

Willmac16, 2020-11-12:
Thank you for reporting the issue. We actually have the fix ready to go and should release it soon.

Willmac16, 2020-11-12:
The new release should fix the issue.

j7126, 2020-11-12:
Sorry about these issues introduced in the update @jlogan223 @Lefuneste83. Hopefully they should be fixed by the 1.16.3 patch that @Willmac16 released. 

I will be testing new updates more thoroughly in the future. 

If any issues persist after installing the patch, please don't hesitate to reply in this thread.

Willmac16, 2020-11-13:
> (Do we have any instructions on how to set up the command widgets? and an example of some code, such as toggling a GPIO state?)
At this point command widgets are only meant to display the result of a unix style command (e.g. to display system uptime you could do something like `uptime | awk '{print $3,$4}' | sed -e 's/,//g'`).

In a future version of Dashboard we may try to implement an active button system (like the option to trigger some enclosure plugin buttons), but for now the command widgets should only receive data since they run on an interval.

jlogan223, 2020-11-13:
@j7126 don't worry, these things happen, happy to find problems if any and report them. Grateful you are all doing this in the first place. @Willmac16 ah ok no problem, I misunderstood its purpose. Thanks for clarifying.

StefanCohen, 2020-11-13:
>  (Do we have any instructions on how to set up the command widgets? and an example of some code, such as toggling a GPIO state?)

There is a section in the wiki: https://github.com/j7126/OctoPrint-Dashboard/wiki/Help#command-widgets

Willmac16, 2020-11-14:
If everything is working for you then I will close this issue.

closed 2020-11-16 by Willmac16