j7126/OctoPrint-Dashboard #238: show the current vs target on the chart

gute1, 2020-11-20:
right now the chart show the target vs the maximum.
you can use different color or extra line to show the current value and then it will be shown against the target and the maximum

Willmac16, 2020-11-21:
Could you expand upon which chart you are referring to? If you are talking about the temperature gauges, they display the target, current, and maximum temperatures all in one gauge.
[image]

Willmac16, 2020-11-25:
@gute1 could you explain which chart you are talking about.

j7126, 2020-12-03:
@gute1 can you please elaborate on what you are asking?

Maybe sending a screenshot would help us better understand your request?

j7126, 2020-12-04:
Closing due to inactivity

closed 2020-12-04 by j7126