j7126/OctoPrint-Dashboard #241: Webcam feed fails

Vaakefjell, 2020-12-04:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
When enabling webcam in Dashboard and making it full screen, after a while the stream becomes unavailable. Affects all services using the webstream: Octoprint Control tab, The Spaghetti Detective, AstroPrint, OctoPod, OctoClient.

**To Reproduce**
Steps to reproduce the behavior:
1) Start a print
2) With other services enabled, go to the dashboard tab with webcam enabled
3) Make full browser window or full screen
4) Wait 15 minutes
5) Webcam freezes and becomes unavailable

**Expected behavior**
I expected the webcam feed to continue.

**Screenshots**
NA
**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log
[octoprint.log](https://github.com/j7126/OctoPrint-Dashboard/files/5645627/octoprint.log)

* Browser console log
* gcode file that can be used to reproduce the problem

**OctoPrint server and Plugins:**
 - Platform: RPi 4B 1.1
 - OS and version: 
 - OctoPrint version: 1.5.0
 - OctoPi version: Linux octopi 5.4.72-v7l+ #1356 SMP Thu Oct 22 13:57:51 BST 2020 armv7l
 - Python version: 
 - Dashboard version 1.16.3
 - DisplayLayerProgress version 1.24.0
 - Access Anywhere - The Spaghetti Detective (1.4.4)

browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/14.0.1 Safari/605.1.15
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: true
connectivity.enabled: true
connectivity.online: true
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: true
env.hardware.cores: 4
env.hardware.freq: 1500
env.hardware.ram: 1903980544
env.os.bits: 32
env.os.id: linux
env.os.platform: linux2
env.plugins.pi_support.model: Raspberry Pi 4 Model B Rev 1.1
env.plugins.pi_support.octopi_version: 0.17.0
env.plugins.pi_support.throttle_state: 0x0
env.python.pip: 19.2.3
env.python.version: 2.7.16
env.python.virtualenv: true
octoprint.safe_mode: false
octoprint.version: 1.5.0
printer.firmware: Marlin Creality 3D

**Desktop (please complete the following information):**
 - OS and version: Issue when using both web browser on RPi and macOS Catalina 10.15.7
 - Browser and version: Firefox on Octopi / Safari 14.0.1 on Mac

**Additional context**
I can't access through VNC for some reason, and no screen at the moment, so my apologies for not providing all the requested information. However, I hope you have what you need to investigate...

Willmac16, 2020-12-05:
@Vaakefjell, I talked to some Spaghetti Detective people, and they recommend switching your premium streaming to compatibility mode. This should let us determine if TSD streaming is the component that is breaking in this specific situation or if the issue lies elsewhere.

Vaakefjell, 2020-12-05:
Thank you for the fast reply!

I am testing it now, I was just about to start a print when I received the
email :)

Another question: When I use the Dashboard plugin in full browser window /
full screen, I can’t see all the information. Is there any way to adjust
the layout? I have tried zooming in the browser, but it makes it look
completely out of proportions and still doesn’t show all...

Same for safari and chrome on Mac, as well as Firefox on the raspberry pi..

Otherwise I really love the plugin, I mostly use it in “ordinary mode”
anyways (with the Cura thumbnail plugin it looks really good :) ), but I
like the looks of the full screen too and would like to explore it some
more.

Best regards,
Hans Harald Berge

Willmac16, 2020-12-05:
You should be able to enable and disable certain widgets for the full screen mode in settings, but we are focusing on a more substantial fix to layout and sizing in the v2 rewrite.

Vaakefjell, 2020-12-05:
Okay!

I’m looking forward to it! Thanks for an amazing job - so far there doesn’t
seem to be any issue with the stream, so most likely it is due to the TSD
premium streaming... I did have a suspicion, but it is always a bit
difficult to know where to start troubleshooting.

Now the TSD stream is choppy though. But as long as the detective works,
and it is up to date, it is fine by me..

Thanks and regards,
Hans Harald Berge

Willmac16, 2020-12-06:
You could create an issue on [TSD GitHub](https://github.com/TheSpaghettiDetective/OctoPrint-TheSpaghettiDetective/issues) (something like "Multiple /webcam/?action=stream streams break premium streaming") and attach the log file.

You may also want to disable filament-sensor-ng debug output if you don't need it since it makes debugging other things slightly harder and some people say that frequent logging can harm your SD card.

closed 2020-12-06 by Willmac16

Vaakefjell, 2020-12-06:
Thanks!

I’ve just forgotten to torn debugging off for the filament sensor.. it was
a pain to get it working reliably 😂

Since I’ve already made a GitHub account, I’ll head over to TSD there (I’ve
been in contact with them on Discord, but that is not really a good way to
register bugs).

Thanks again, and have a nice day!
Hans Harald Berge