j7126/OctoPrint-Dashboard #242: Webcam ignoring Flip options on startup

help137, 2020-12-11:
**Describe the bug**
When you first logon to Octoprint (Dashboard is set as my first tab) the Flip options on the webcam setup are ignored. In my case this means the webcam view is upside down and reversed as I use both Flip Horz & Vert options in my setup. You have to close and reopen the webcam view with the button in the upper right corner before it displays properly. 

I've also noticed that the webcam view sometimes doesn't update at all when starting a print job (it's just frozen on a still frame) until the same open/close has been done. This last issue only seems to have started with version 1.16.3 of the plugin.  

**To Reproduce**
Steps to reproduce the behavior:
1. Turn on the Flip options in the Octoprint Webcam config
2. Logon to Octoprint. 

**Expected behavior**
The webcam view should be obeying the Flip options on startup.

**Files (please attach the following Files or screenshots when applicable):**
N/A

**OctoPrint server and Plugins:**
 - Platform: RPi 3A+
 - OS and version: Linux octopi 4.19.75-v7+
 - OctoPrint version: OctoPrint 1.4.2
 - OctoPi version: OctoPi 0.17.0
 - Python version: Python 2.7.16
 - Dashboard version Dashboard: 1.16.3
 - DisplayLayerProgress version: 1.24.0

**Desktop (please complete the following information):**
 - OS and version: Windows 10
 - Browser and version Firefox 83

j7126, 2020-12-14:
Hi, @help137 can you please upload your octoprint log file

help137, 2020-12-16:
> 
> 
> Hi, @help137 can you please upload your octoprint log file

Here you go. This is today's logon, the camera was upside down/backwards as usual. I clicked the button in the top right to toggle it off and back on, then downloaded the log. The only thing that has been altered about the log is a search and replace on my user name was done to replace it with USERNAME. 

-K
[octoprint.log](https://github.com/j7126/OctoPrint-Dashboard/files/5705570/octoprint.log)

Willmac16, 2020-12-16:
I was able to reproduce the bug on my end, so I will attempt to fix the issue.

ChuckRider, 2020-12-24:
I am seeing the same thing.  Just searched to see if there was any mention of it.  I am willing to beta test the solution if you like.  Running on rPi 3B with latested OctoPi and plugin updates.

Willmac16, 2020-12-24:
If you can, install in the plugin manager with this link: `https://github.com/j7126/OctoPrint-Dashboard/archive/master.zip`

ChuckRider, 2020-12-24:
Working.  Thank you!

Willmac16 wrote on 12/24/2020 1:27 PM:

charredchar, 2020-12-29:
Attempted the development zip and it does not resolve my issue, same problem. Rotation works great but I need a 270 degree rotation, or in other words rotate, flip horizontal and vertical. Works in OctoPrint's Control tab.

ChuckRider, 2020-12-29:
It fixed mine

charredchar wrote on 12/29/2020 12:52 PM:

>
> Attempted the development zip and it does not resolve my issue, same 
> problem. Rotation works great but I need a 270 degree rotation, or in 
> other words rotate, flip horizontal and vertical. Works in OctoPrint's 
> Control tab.

<https://www.postbox-inc.com>

Willmac16, 2020-12-29:
@charredchar I cannot reproduce your issue. If you have screenshots or more information that you can share, please do so in a new issue.

charredchar, 2020-12-30:
> It fixed mine

That's good for you...

> @charredchar I cannot reproduce your issue. If you have screenshots or more information that you can share, please do so in a new issue.

I'd figured you'd want to consolidate same issues considering this one was still open. But whatever, thanks for the help, with the other issues I've had I'll just remove the plugin.

j7126, 2020-12-30:
> I'd figured you'd want to consolidate same issues considering this one was still open. But whatever, thanks for the help, with the other issues I've had I'll just remove the plugin.

Sorry for the bad experience you have had width the plugin. I would like to help resolve the issues you have had. If the fix in the development branch does not work for you, please upload your octoprint log file here and I will try to resolve your issue. Please feel free to let us know of any other issues you have had.

j7126, 2020-12-30:
> I'd figured you'd want to consolidate same issues considering this one was still open. But whatever, thanks for the help, with the other issues I've had I'll just remove the plugin.

@charredchar 

Sorry for the bad experience you have had width the plugin. I would like to help resolve the issues you have had. If the fix in the development branch does not work for you, please upload your octoprint log file here and I will try to resolve your issue. Please feel free to let us know of any other issues you have had.

help137, 2020-12-31:
The 1.17 Dev version appears to have fixed this issue for me as well. Thank you very much.

closed 2020-12-31 by help137

charredchar, 2021-01-25:
I decided to give it another shot when I saw the update notes stating it was fixed, still a problem for me. Flip Horizonal, Vertical and 90CCW seem to work just fine independently but attempting to rotate 90 degrees and flip vertically/horizontally (needed if you want to rotate clockwise) does not work.

Willmac16, 2021-01-26:
@charredchar I was able to reproduce the issue and will start working to fix it.

j7126, 2021-01-26:
@charredchar I have fixed your issue in the development branch. Please install `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip` to confirm that the fix works for you.