j7126/OctoPrint-Dashboard #243: Fullscreen & Full Browser Issue with UI Customizer.

Pheelix, 2020-12-14:
I don't this this is a issue with your plugin but I just wanted to let you know that there is a overlay issue with Fullscreen modes and the new UI Customizer plugin.  I don't know if you want to talk to the creator of UI Customizer to figure out a fix or just let them figure it out, as it is there plugin that is overlaying on Dashboard.

Here is my Issue post in the UI Customizer Repo.
https://github.com/LazeMSS/OctoPrint-UICustomizer/issues/32

Picture of the Issue.
[image]

Pheelix, 2020-12-14:
@j7126 You can close this unless you need more info or need me to tests things on my side for you.  My findings so far were posted on the UIC post I made.

j7126, 2020-12-14:
I have made a fix, it will be in the next release.

closed 2020-12-14 by j7126

Pheelix, 2020-12-15:
@j7126 Just tested the fix and it is working properly now with the theme colors checked.  Thanks