j7126/OctoPrint-Dashboard #244: Consider Adding Setpoint Details to Dial

InoSiX, 2020-12-16:
Love this simplified interface. One thing I think it lacks is context. There is no min/max numbers on either end of the dial style bar graphs. 

Additionally, there is no indication as to where the temperature is currently heading to. The bar fills up with current temperature but does not indicate where it intends to stop (it's current temperature setpoint). This would be valuable to provide context to the user. To keep with the simplified design, a line cutting through the dial at the current setpoint or perhaps a triangle along the outer line with a 'SP' for setpoint?

Willmac16, 2020-12-16:
Could you send a screenshot of what you see on on your end (just a screenshot showing the temp dial should be enough). I want to make sure there isn't some theme color visibility conflict.

Willmac16, 2020-12-16:
With my theme colors it looks like this: https://user-images.githubusercontent.com/27445690/99888412-7b4a7a00-2c01-11eb-8d03-4f793bf9bd66.png

Willmac16, 2020-12-16:
@InoSiX could you send a screenshot of what you see on on your end (just a screenshot showing the temp dial should be enough). I want to make sure there isn't some theme color visibility conflict.

InoSiX, 2020-12-16:
I have uploaded a screenie when I am connected and a print is started (bed would be heating in this screenie).

See: https://i.imgur.com/Wl7Z7PE.png

Willmac16, 2020-12-16:
It looks like the plugin isn't registering the target temp at all. This is what it looks like on my end when I heat up my bed.
[image]

To investigate this further could you share your printer model and tell me if the target temps show up in the temperature tab of OctoPrint.

InoSiX, 2020-12-16:
Ahhhh, you are refering to the Target: 60°C in the center of the picture. Yes, I am aware that exists. And it normally functions (to take the screenie, I turned on my printer, connected, hit print on something and immediately took the screenie. This number normally updates).

The purpose of the dial is to provide an at-a-glance reference for current conditions. The dial as it is designed currently does not provide any context to any conditions. It's exactly arbitrary. Although the user can supply the dial min/max values, the user must remember these values when evaluating the bar fill. Further, the target temperature is not captured in the bar at all.

I don't say this with complete lack of experience. I literally design industrial (oil/gas/chemical/power) HMI/GUI as a large part of my day to day work. Gimme a few, I'll mark something up really quick.

StefanCohen, 2020-12-16:
Hi. I'm pretty sure that Willmac16 is refering to the grey bar in the middle of the gauge that indicates the target temperature.

InoSiX, 2020-12-16:
Wow, I ran an errand and sat back down to edit this before upload, While doing that, I was about to use the gray from that internal dial when I realized what that darker gray color was attempting to indicate. Thank you for pointing that out.

InoSiX, 2020-12-16:
I made a simple edit to add scaling on the dial. It probably would be better served with additional demarcations for things like 25, 50, 75% EU, similiar to traditional speed indicators.

[image]

Willmac16, 2020-12-16:
That looks great! Just to check, does the existing target temp line render for you?

InoSiX, 2020-12-16:
Yes, it does. It doesn't really jump out at you though. It's easy for me to see how I manage to miss it. Perhaps something to make it more visible or approach it a different way?

Willmac16, 2020-12-25:
[image]
I still need to ensure it is theme safe, but this is an active working version of it.

ChuckRider, 2020-12-25:
> image

Nice.  Let me know if you want me to beta test it.

Willmac16 wrote on 12/24/2020 8:34 PM:

Willmac16, 2020-12-25:
`https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`

ChuckRider, 2020-12-25:
I loaded it and it ran fine.  Love the new look.  I then power cycled 
the printer and could not reconnect.  I tried multiple combinations of 
reboot the rPi and the printer but nothing brought it back.  I disabled 
Dashboard in OctoPi and it everything started working.  I am running 
Klipper firmware if that makes any difference.  This is the messages in 
the terminal window on restart:

Unexpected error while reading serial port, please consult octoprint.log 
for details: SerialException: 'read failed: device reports readiness to 
read but returned no data (device disconnected or multiple access on 
port?)' @ comm.py:_readline:3831 Please see 
https://faq.octoprint.org/serialerror for possible reasons of this. 
Changing monitoring state from "Operational" to "Offline (Error: 
SerialException: 'read failed: device reports readiness to read but 
returned no data (device disconnected or multiple access on port?)' @ 
comm.py:_readline:3831)" Connection closed, closing down monitor

Are there any log files that I need to capture and send you?

Willmac16 wrote on 12/25/2020 12:22 AM:

Willmac16, 2020-12-25:
@ChuckRider, the `~/.octoprint/logs/octoprint.log` file is the one; however, that issue is most likely something other than this plugin, especially since we haven't changed much at all of the server side of the plugin recently, so I would suspect it is either that you accidentally tried to run a second instance of octoprint or some hardware issues.

ChuckRider, 2020-12-25:
Seems to be related to the jog control button option.  If I move it to 
the disabled list then everything works.

Observations:
1) It takes about 15 seconds to reload OctoPrint without the Dashboard 
installed.  It takes about 90 seconds with it.
2) There is about a 45 second delay in the webcam when Dashboard is 
installed.  This occurs even in the 'Command' tab, not just the 
Dashboard tab so I think it may be a something outside of Dashboard per 
se but aggravated by it.

Printing in black PETG so it is hard to see on the webcam but here is a 
screenshot with things running.

Willmac16 wrote on 12/25/2020 10:24 AM:

ChuckRider, 2020-12-25:
Here is my log file

Willmac16 wrote on 12/25/2020 10:24 AM:

Willmac16, 2020-12-29:
@ChuckRider could you send the files again. I believe that they did not go through in the email reply.

j7126, 2020-12-30:
I noticed that if you change the number of ticks to something other than the default, when you load the page it is back to the default and only changes to what you set if you open settings and close them again.

Willmac16, 2020-12-31:
Ready for Next Release.

closed 2021-01-11 by j7126