j7126/OctoPrint-Dashboard #249: Plans / Feature requests for v2

j7126, 2021-01-11:


cp2004, 2021-01-12:
We're all watching this one closely with a new core UI in mind, look forward to seeing it happen.

Good luck!

jneilliii, 2021-01-12:
definitely watching...I know Charlie has been playing with replacing components in the default UI with react pieces as a learning process.

jrmoserbaltimore, 2021-05-05:
I'd like to see a few things.  The temperature graph is a definite one if you're replacing the UI.

I actually would like to see a graph of TX speed, especially if integrated with MeatPack.  Currently it just divides the entire amount sent by the entire print time, so when you wait 10 minutes to heat up and level the bed in an 8 minute print you get a pathetically-low TX measure.  Sometimes I see 2k pass right by in one second, but it says 173 bytes/sec.  Using the simple moving average over the past *n* seconds, for some value of *n*, would work for the dial-style readout of current TX speed.

exotolon, 2021-05-09:
Great work with v1 of your plugin!
I use the dashboard alot in fullscreen - especially as it gives me a quick overview over erverything including a webcam view of what's going on (the printer is in a seperate room).

I'd love to be able to hde the filename - which is irrelevant info for me. Otherwise the number of (total) layers and (final) height is cut off.

JayRen78, 2021-07-30:
I had made a thread in requests.  But I'd love to see something like this in 2.0

[image]

SaKiEQ, 2021-11-07:
I think it would interesting to have the ability to have more than one Dashboard config at a time. 
Sort of like having Dashboard tabs.  Different views for different purposes, sort of like it is implemented
right now for the different function areas, like motion, heating, etc... so that a selected Dashboard set
can be made full screen. 

On some computers I might want to have a full screen view of the basics - progress, temps, camera feed. 
While on others it might be more machine centric, with all the details needed. 

I very much would like to have the icon / preview of the model on the full screen view. 
A picture of the model printing often tell a much quicker story of what is going on, then text 
and labels.  

I am running multiple machines, with each its own octoprint version... I am using themefy to color code the printers so the UI in the browser is easy to distinguish when switching between browser tabs. I find themefy a bit cumbersome to configure, maybe there could be a better way of selecting basic color schemes or more refined schemes than the current ones.

DrEthan77, 2022-08-10:
I just submitted a feature request but I would love to see webrtc implemented!