j7126/OctoPrint-Dashboard #250: Display ETA in 12 or 24 hour format

Jeff-Haas, 2021-01-13:
Current ETA is in 24 hour format:

[image]

Add config option to allow 12 hour format with AM/PM.

devin122, 2021-01-19:
I was just looking at hacking this in myself.  For reference the relevant bit of code is [here](https://github.com/j7126/OctoPrint-Dashboard/blob/35a76611ed00ff0d4e448ceee2c982404a9942a3/octoprint_dashboard/static/js/dashboard.js#L550) .  I think the best solution would be to use [`toLocaleTimeString()`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Date/toLocaleTimeString) instead of [`toTimeString()`](https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Date/toTimeString), since it will default to 12/24hr as appropriate for the users locale. On top of that you could add a settings  to force 12/24hr format (also maybe an option to override  the default locale?)

mbourque, 2021-02-08:
Looking forward to this fix.

closed 2021-03-15 by j7126