j7126/OctoPrint-Dashboard #253: No cancel print dialog in full browser/screen mode

th3obituary, 2021-02-09:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
I was starting a print and watching the first layers going poorly and decided to hit cancel but it did not then i thought oh yeah you forgot to click cancel or continue on the dialog box that pops up asking if your sure you want to cancel then tried again only to notice it was not popping up until i left full browser mode.
I tried this in both full browser window mode and fullscreen mode on:

**To Reproduce**
Steps to reproduce the behavior:
1. start a print
2. click either the full browser or full screen button in dashboard tab
3. Click cancel

**Expected behavior**
expect the normal cancellation.
**Screenshots**
don't help as its going to look normal, abnormality is what is missing

**Files (please attach the following Files or screenshots when applicable):**
* Octoprint.log is something i would have to get back with you on although it should be easy to reproduce
* Browser console log (i would have to look up, not something i have to mess with often
* should work with any gcode file

**OctoPrint server and Plugins:**
 - Platform: Rpi 4b 2gig vesion
 - OS and version: octopi? first Pi i have owned and only installed octopi/print on it
 - OctoPrint version: OctoPrint 1.5.3
 - OctoPi version: OctoPi 0.18.0
 - Python version Python 3.7.3
 - Dashboard version 1.17.0
 - DisplayLayerProgress version 1.25.3

**Desktop (please complete the following information):**
 - OS and version: Windows 10 Pro 64bit OS build19042.74
 - Browser and version 

[Firefox (85.01 64bit)
Brave (Version 1.19.92 Chromium: 88.0.4324.152 (Official Build) (64-bit))
Opera (Version:73.0.3856.344)
Edge  (Version 88.0.705.63 (Official build) (64-bit))

**Logs**
Please upload your OctoPrint log file 
said it was too big (36mb)
**Additional context**
I am very new to printing having a cr6se for about a month and a Raspberry Pi only about 2 weeks so very new to both those and only can accomplish that likely due to beind ok with computers and A= certified. so forgive my lack of knowledge in the 2 areas in new in and sorry in advance for and dumb questions/oversights on my part. Thank you for your work on this plugin!!

th3obituary, 2021-02-09:
Forgot!! Out of the 4 tested browsers all acted the same except i noticed with brave the confirmation dialog seemed to pop up but hidden until i left full browser or fullscreen mode. The other 3 i only seen it after returning to normal view and hitting cancel again.

j7126, 2021-02-09:
I have made a fix. It will be included in the next release

[image]

th3obituary, 2021-02-09:
Awesome! keep up the great work

closed 2021-03-15 by j7126