j7126/OctoPrint-Dashboard #255: HLS Webcam urls do not work

raiscan, 2021-02-11:
Hi!

HLS streams currently do not display for webcams, although they are supported by octoprint 1.5.x (think it's recent). When a HLS stream url is used, a broken page icon is displayed.

Steps to reproduce the behavior:
1. (Assuming a Pi running OctoPi 0.18, OctoPrint 1.5.3 and Dashboard 1.17.0) enable camera_streamer=hls at the bottom of /boot/octopi.txt
2. Change stream url in octoprint to /hls/stream.m3u8
3. The stream url will work correctly for octoprint (test will work, "Control" page will work), but not under the dashboard.

Willmac16, 2021-02-12:
I got HLS working (at least with a test stream from the internet). You can try it out by installing with the url `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`

Willmac16, 2021-02-12:
If you happened to try out the dev branch already, I have made some more changes to ensure webcam streams are only running when necessary, so you should try the new version.

raiscan, 2021-02-12:
First off, thanks so much for being so prompt! I regret I was going to message earlier but spinning plates!

I have redownloaded the dev archive.

Can confirm this works great :) *however* It took a browser refresh from me entering the correct stream though (can the plugin recheck if hls on url change?). 

Otherwise thanks so much!

Willmac16, 2021-02-12:
Nice catch! I fixed that behavior int the latest commit. Thank you for you help.

closed 2021-02-14 by Willmac16