j7126/OctoPrint-Dashboard #257: Enclosure Plugin Temperature and Humidity Control Widget with V2?

wickedbeernut, 2021-02-15:
The Dashboard plugin currently supports Enclosure plugin rpi_inputs of type temperature_sensor (e.g., enclosureViewModel.rpi_inputs_temperature_sensors()).  If enabled, the Dashboard plugin displays a widget with the measured temperature and humidity.

From “Plans / Feature requests for v2”, _Additionally, an interface will be made available to allow other plugins to add widgets and layouts to the dashboard interface._

The Enclosure plugin also supports rpi_outputs of type temp_hum_control.  In addition to the measured temperature (and optionally measured humidity), the temp_hum_control output also includes a target temperature.

Will the v2 interface allow the Enclosure plugin add a widget to the dashboard with measured temperature, measured humidity and target temperature?

[image]

j7126, 2021-02-21:
Yes this will be possible with v2

wickedbeernut, 2021-02-21:
> @j7126 said, "Yes this will be possible with v2".

Excellent!

I edited the issue to read more like a feature request than a question.

j7126, 2021-02-27:
The thing with v2 will be that almost anything will be possible, as integrations with other plugins will be more up to that plugin developer. 

There will also be much more freedom for the user to customize widgets, as the idea will be that the user will be able to add a widget, choose it's type (like gauge, text progress bar, button etc.) and then choose a data point or action from OctoPrint or any plugin to apply to that widget.

j7126, 2021-02-27:
I am closing this issue now as i would like to keep v2 related discussion in #249 for now.

closed 2021-02-27 by j7126

wickedbeernut, 2021-03-11:
@j7126, I understand with v2 anything will be possible, but I just wanted to share one more mockup.

This is what my six temperature controllers (heaters and coolers / fans) corresponding to three filament dryers actually look like in the Enclosure plugin (this is not a mockup),

[image]

I plan to leverage the Dashboard plugin v2 to display the six temperature controllers as,

[image]