j7126/OctoPrint-Dashboard #259: Commands not working.

delurfangs, 2021-03-06:
I know that this is a bit long but please take your time and fill in the following information as clearly and detailed as possible. It will shorten the time needed to understand and fix the issue significantly. Please also state versions clearly when applicable. "Latest version" doesn't help much.  

**Describe the bug**
Commands fail to run and produce no output 

**To Reproduce**
Steps to reproduce the behavior:
create and enable a command and allow it to attempt to run by set interval or the test button 

**Expected behavior**
command to run 

**Screenshots**

**Files (please attach the following Files or screenshots when applicable):**
2021-03-05 22:08:05,832 - octoprint.plugins.dashboard - INFO - testCmdWidget called, cmd is echo hello
2021-03-05 22:08:05,875 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin dashboard
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 154, in on_api_command
    self.testCmd(data["cmd"])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 112, in testCmd
    except subprocess.TimeoutExpired:
AttributeError: 'module' object has no attribute 'TimeoutExpired'
2021-03-05 22:08:05,893 - tornado.access - ERROR - 500 POST /api/plugin/dashboard (::ffff:192.168.29.57) 82.19ms

**OctoPrint server and Plugins:**
 - Platform: RPi 3b+
 - OS and version: raspbian 10
 - OctoPrint version: 1.5.3
 - OctoPi version: 0.17.0
 - Python version 2.7.16
 - Dashboard version 1.17.0
 - DisplayLayerProgress version 1.25.4
 - 
**Desktop (please complete the following information):**
 - OS and version: windows 10 
 - Browser and version chrome 88.0.4324.190 (Official Build) (64-bit)

Willmac16, 2021-03-06:
Looks like Dashboard needs to not try to except that error in versions before 3.3. You can try the patch by installing from the url `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`, but I would recommend that you upgrade to python 3 at some point since python 2 is end of life.

https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

delurfangs, 2021-03-06:
I am in the middle of a 14 hour print right now and with work I will not be able to do anything with it, but when I can I will update to python 3, I did not even realize until was filling this out that it was still on 2 but if you would like me to test that patch first I am more than happy to.

Willmac16, 2021-03-06:
If you are willing to that would be great, but if you don't find the time I can run the test on a octopi I have still running python 2.

delurfangs, 2021-03-06:
No problem I will test it and let you tomorrow around 7pm or 8pm central time.

delurfangs, 2021-03-07:
still happens with the development zip you linked but it is different lines giving errors now:  
2021-03-06 18:10:22,356 - octoprint.plugins.dashboard - INFO - testCmdWidget called, cmd is echo 9V
2021-03-06 18:10:22,393 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin dashboard
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 157, in on_api_command
    self.testCmd(data["cmd"])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 118, in testCmd
    stdout, stderr = process.communicate(timeout=10.0)
TypeError: communicate() got an unexpected keyword argument 'timeout'
2021-03-06 18:10:22,418 - tornado.access - ERROR - 500 POST /api/plugin/dashboard (::ffff:192.168.29.57) 76.01ms
2021-03-06 18:10:33,182 - octoprint.plugins.dashboard - INFO - testCmdWidget called, cmd is echo hello
2021-03-06 18:10:33,222 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin dashboard
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/server/api/__init__.py", line 146, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/util/__init__.py", line 1890, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 157, in on_api_command
    self.testCmd(data["cmd"])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_dashboard/__init__.py", line 118, in testCmd
    stdout, stderr = process.communicate(timeout=10.0)
TypeError: communicate() got an unexpected keyword argument 'timeout'
2021-03-06 18:10:33,246 - tornado.access - ERROR - 500 POST /api/plugin/dashboard (::ffff:192.168.29.57) 85.08ms

I will hold off on installing python3 unless you want to just say commands require python3 which is fair.

Willmac16, 2021-03-07:
I tested the latest version on python2 and python3 and both work, so go for the install if you want to.

delurfangs, 2021-03-07:
Worked for me as well. Thanks for your quick work on it. Now to update python and get out of the stone age.

closed 2021-03-07 by delurfangs