j7126/OctoPrint-Dashboard #260: Feature request: Layer durations graph to show predicted durations

Sophist-UK, 2021-03-11:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Rather than show just the actual durations for layers already printed, also show predicted layer durations for all layers.

**Describe alternatives you've considered**
None.

**Additional context**
Hopefully self explanatory without graphics.

StefanCohen, 2021-03-12:
That's a really challenging feature request :) the printtimegenius plugin does a pretty good job of estimating the total print time (at least for the Marlin firmware) but doesn't provide estimates per layer afaik.

Sophist-UK, 2021-03-12:
Could we use the comments in the gcode for this (even if the total of the individual layer estimates is different from PrintTimeGeniius)?

Willmac16, 2021-03-12:
We could look into using the same backend estimation as PrintTimeGenius while processing one layer at a time, but I cannot imagine another good solution since not all slicers put a time estimate per layer into the gcode.

Sophist-UK, 2021-03-12:
If the slicer doesn't put time estimates into the gcode, then you cant do it. But if the slicer does, then it would be great to see actual vs. estimated.