j7126/OctoPrint-Dashboard #262: Layer Percentage not updating

drewstopherlee, 2021-03-15:
Describe the bug:
The layer percentage does not change and stays at 0%. The layer durations, job percentage, and all other data outputs update normally.

To Reproduce Steps to reproduce the behavior:
1. Open OctoPrint in browser.
2. Open a local .gcode file.
3. Begin print.
4. Once print begins, the layer percentage graph does not move.

Expected behavior:
The layer percentage graph updates dynamically based on layer completion (as it used to about a week ago).

Screenshots:
[image]

[image]

Files:
- Octoprint.log
[octoprint log.log](https://github.com/j7126/OctoPrint-Dashboard/files/6138784/octoprint.log.log)

- Browser console log
[chrome console.log](https://github.com/j7126/OctoPrint-Dashboard/files/6138788/chrome.console.log)

- gcode file that can be used to reproduce the problem
[PegboardTowelHolder 0.2mm PLA 16h55m.gcode.zip](https://github.com/j7126/OctoPrint-Dashboard/files/6138800/PegboardTowelHolder.0.2mm.PLA.16h55m.gcode.zip)

OctoPrint server and Plugins:
- Platform: RasPi 4B, 4GB RAM (Rev 1.2)
- OS and version: Raspbian GNU/Linux 10 (buster)
- OctoPrint version: OctoPrint 1.5.3
- OctoPi version: OctoPi 0.18.0
- Python version: Python 3.7.3
- Dashboard version: 1.17.0
- DisplayLayerProgress version: 1.25.4

Desktop:
- OS and version: macOS Big Sur (11.1)
- Browser and version: Chrome (Version 89.0.4389.82)

Logs:
[octoprint.log](https://github.com/j7126/OctoPrint-Dashboard/files/6138802/octoprint.log)

Additional context:
The only recent change (past two weeks) to my OctoPrint instance has been switching from the FilamentManager plugin to the SpoolManager plugin.

Willmac16, 2021-03-15:
The current version of Dashboard uses the gcode viewer plugin to display layer progress, and it will not automatically load large files on the front end. Most likely, you loaded a file above the limit that gcode viewer has set. This will not be a problem in future versions of Dashboard that use the backend to track layer progress.

Willmac16, 2021-03-15:
If you want to try an early version, you can install with `https://github.com/j7126/OctoPrint-Dashboard/archive/dlpreplacement.zip`

drewstopherlee, 2021-03-15:
Thanks for the quick reply, I think this is exactly what's going on. When I try to view the gcode using gcode viewer, it displays a warning regarding file size and I have been printing larger files recently. I'll try that install you mentioned. Thanks!

closed 2021-03-15 by drewstopherlee

drewstopherlee, 2021-03-17:
>`https://github.com/j7126/OctoPrint-Dashboard/archive/dlpreplacement.zip`

When I installed using this link, it still wasn't working. I chose the release candidate channel in the Software Update menu and re-installed, and now it's working as expected. Thanks again for your help!

Willmac16, 2021-03-17:
@drewstopherlee just to be clear, the release candidate channel will install version 1.18.0 of Dashboard (the latest release), not the (unreleased) version with Dashboard backend file processing, so you will likely run into the same issue with large files (you can always manually go into the gcode viewer and force them to load).

It is possible that your slicer'a layer markers are not recognized by Dashboard, so if you are willing to share the file that it did not work on, that would be helpful to ensure the feature works for more users in the future.

drewstopherlee, 2021-03-17:
Huh, okay, I didn't know that. It must've been the size of the file then, the dlpreplacement.zip install gave me the same issue when I ran the same .gcode file a second time. 
Here's a zip of the gcode that didn't load layer progress: 
[PegboardTowelHolder 0.2mm PLA 16h55m.gcode.zip](https://github.com/j7126/OctoPrint-Dashboard/files/6159203/PegboardTowelHolder.0.2mm.PLA.16h55m.gcode.zip)
And here's the gcode for a file that _did_ display layer progress:
[Calipers 2x #1 0.2mm PLA 2h56m.gcode.zip](https://github.com/j7126/OctoPrint-Dashboard/files/6159214/Calipers.2x.1.0.2mm.PLA.2h56m.gcode.zip)

Willmac16, 2021-03-17:
Awesome, I'll try those files out, and we will likely run a Release Candidate for the backend rework update.

drewstopherlee, 2021-03-17:
Thanks for your insight! I really appreciate it.

Willmac16, 2021-03-18:
Okay, I took a look, and the files seem to have the same layer markings, so I bet that you uploaded the first file before upgrading to the test version and uploaded the second file afterwards. That specific version of Dashboard processes upon file upload and doesn't try to process on file print.

drewstopherlee, 2021-03-19:
That's definitely it, I didn't know they processed on upload. I haven't had any similar issues with new files I've uploaded. Thanks for checking that out.