j7126/OctoPrint-Dashboard #263: Webcam state is not honored when moving away from the dashboard

samwiseg0, 2021-03-16:
**Describe the bug**
In version prior to 1.18.0 when moving to another tab and then moving back to dashboard the webcam state was honored. Since the changes to how the webcam is loaded this is no longer honored. This also seems to be a problem when moving away from the browser tab as well. 

Additionally, this may fall under another request/feature request, but it would be nice if the state is also pervasive on page reloads as well. 

**To Reproduce**
Steps to reproduce the behavior:
1. Load octoprint UX
2. hide the webcam from the dashboard
3. Move to another browser tab or octoprint tab
4. Move back to the dashboard tab
5. Observe that the webcam is now visible. 

**Expected behavior**
The state of the webcam should be honored and not changed when moving to other browser tabs and/or octoprint tabs. 

**Screenshots**

[image]

**Files (please attach the following Files or screenshots when applicable):**
NA

**OctoPrint server and Plugins:**
 - Platform: **Rpi 4B**
 - OS and version: **Raspbian**
 - OctoPrint version: **OctoPrint 1.5.3**
 - OctoPi version: **OctoPi 0.18.0**
 - Python version **Python 3.9.0**
 - Dashboard version **Dashboard 1.18.0**
 - DisplayLayerProgress version **1.25.4**

**Desktop (please complete the following information):**
 - OS and version: MacOS
 - **Chrome Version 89.0.4389.82 (Official Build) (x86_64)**, **Safari Version 14.0.3 (16610.4.3.1.7)**

**Logs**
NA

**Additional context**
NA

Willmac16, 2021-03-16:
Thanks for reporting this. I have fixed the issue and you can try out the fixes by installing with `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`

samwiseg0, 2021-03-16:
Confirmed that is works correctly now! Thanks for fixing it so fast!

Willmac16, 2021-03-18:
Fix was released with 1.18.1

closed 2021-03-18 by Willmac16