j7126/OctoPrint-Dashboard #265: Dashboard starts streaming camera even with Widget Disabled

paulo-serrao, 2021-03-17:
**Describe the bug**
Even with the camera widget disabled and not showing on the page, the dashboard automatically starts streaming the camera feed on the background using bandwidth.
On the widget options, deleting the camera feed does not solve the problem, I could only stop the streaming by:

Enabling the Widget and manually clicking the camera icon to hide/stop the camera feed.
Disabling the Widget and selecting "Enable Dashboard Multi Webcams" while having no camera feeds.

**To Reproduce**
Load the web ui
Check network card bandwidth usage

**Expected behavior**
Don't start camera streaming if widget is disabled

**OctoPrint server and Plugins:**
    Platform: Raspberry Pi 4 Model B Rev 1.2
    OS and version: OctoPi 0.17.0
    OctoPi version: OctoPi 0.17.0
    Dashboard version: Dashboard 1.18.0

**Desktop (please complete the following information):**
    OS and version: macOS Catalina
    Browser and version: Firefox 86.0.1

Willmac16, 2021-03-17:
Thanks for catching that. You can try the fix by installing with the url `https://github.com/j7126/OctoPrint-Dashboard/archive/development.zip`

Willmac16, 2021-03-18:
Fix pushed with 1.18.1

closed 2021-03-18 by Willmac16

paulo-serrao, 2021-03-18:
I confirm 1.18.1 no longer have the issue :)